cotestatnt/AsyncTelegram2 #1: Please reduce the number of Serial.println ();

nvjob, 2021-04-06:
Please reduce the number of Serial.println ();, now this is very much clogging up the serial port.

nvjob, 2021-04-06:
#define DEBUG_ENABLE 0 - it doesn't help. You will have to delete everything yourself.

nvjob, 2021-04-06:
Changed DEBUG_ENABLE 1 to DEBUG_ENABLE 0 in AsyncTelegram2.h itself, it helped.

nvjob, 2021-04-06:
Please add disabling debug.
For example - AsyncTelegram2 myBot(client, false);

cotestatnt, 2021-04-10:
Hello @nvjob 
Thanks for your comment and support.
Library is still in development, so many lines of unnecessary code are still present, such as Serial.print ()
A stable release clearly will print debug info only if user set manually DEBUG_ENABLE to 1 (default will be 0). 

Keep in mind that if you are using the SSLClient library, a lot of debug information is generated by library itself.
If they are not needed, create the instance like this:
`SSLClient client (base_client, TAs, (size_t) TAs_NUM, A0, 1, SSLClient::SSL_NONE);`

closed 2021-04-23 by cotestatnt