cotestatnt/AsyncTelegram2 #2: ESP8266 Unstable

vlw, 2021-04-16:
1. Sometimes reboots (20 times in day)
2. Sometimes lost connection until manual reboot
3. OTA 8266 cause cyclic reboot

vlw, 2021-04-16:
https://gist.github.com/vlw/29a660f41fe6ed00729dd839b7625d93

log of crash after OTA

vlw, 2021-04-16:
https://gist.github.com/vlw/a887076d4c2ba74c55acff7f0c1c44fe
wifi with poor internet (crash on start)

cotestatnt, 2021-04-17:
Hi Vladimir! Thanks for your great support.
I'm experiencing same issues and my first impression is that this crashes are related to memory management.
In the last commits indeed, I've worked around variables allocation and similar points trying to optimize it.

Reverting back for example to the old commit db95b01, i was able to solve this crashes, but i'm still searching to find the root cause.
Now my plan is to re-do the steps i've done until system become instable again in order to have clean what's happening.

vlw, 2021-04-18:
Hi Tolentino! Tested ESP8266 Ota query sample in state d7de7fb865ff34e2478ceeb7f16216bd708cf590
Now bot not responding on inline kerboard :(
[image]

vlw, 2021-04-18:
if I start OTA update without inline button, I get Soft WDT reset
[image]

vlw, 2021-04-18:
Still random reboots in 30min

cotestatnt, 2021-04-20:
This issues is very annoying, i was not able to circumstantiate the root cause.
I've done so many tests without stable result...
Only reverting back to v1 library approach with a global HTTP rx buffer (String for the moment), seems much more stable with esp8266 (i've uploaded files in dev branch if you want try).

closed 2021-05-17 by cotestatnt