cotestatnt/AsyncTelegram2 #4: Send photo to group possible?

philicibine, 2021-05-22:
Hi, thanks for this very useful library.  I'm using it for a doorbell project.  I am wondering if it is possible to send a photo to a group?  I can't really see how looking through the source.

Thanks in advance.

cotestatnt, 2021-05-24:
Thanks to you for support!
Yes it is possible, but you need to know in advance the **group id**.
Grupo id is a negative integer that uniquely identifies the group chat, but discovering from Telegram side is not so easy unfortunately.
You can use for exampli one of this bots, inviting them to your group [@getidsbot](https://t.me/getidsbot) or [@RawDataBot](https://t.me/RawDataBot).

Once you have the id, sending a message is exactly as for the other type fo chats.

cotestatnt, 2021-05-24:
Maybe the ESP32-CAM example has to be updated :(
I will check as soon as i can.

AlexCrane73, 2021-06-01:
**group id** is much higher than **int32_t** range, **int64_t** type is needed

cotestatnt, 2021-06-03:
Yes, you are right @AlexCrane73 
I'm going to edit all occurrence of int32_t to **int64_t**

closed 2021-06-03 by cotestatnt