cotestatnt/AsyncTelegram2 #5: lost messages

droidprova, 2021-06-03:
Ciao Tolentino, provando l'esempio ESP32-CAM che hai aggiornato qualche giorno fa (ma succedeva anche con le precedenti versioni), noto che se si inviano messaggi molto ravvicivati (qualche secondo) la scheda (ESP-32 CAM) pur ricevendo il messaggio non scatta e di conseguenza non invia la foto. E' come se si perdesse dei messaggi. Idee?

Hi Tolentino, trying the ESP32-CAM example that you updated a few days ago (but it also happened with the previous versions), I notice that if you send very close messages (a few seconds) the card (ESP-32 CAM) while receiving the message does not take and consequently does not send the photo. It's like missing messages. Ideas?

cotestatnt, 2021-06-03:
Ciao @droidprova 
Ci sono due esempi per ESP32-CAM, a quale ti riferisci? Con qualche secondo intendi più 2/3 secondi o 10/20?
Considera che l'upload del file sul server Telegram è piuttosto lento [(è un tema piuttosto caldo con l'ESP32](https://www.google.com/search?q=esp32+wificlient+upload+slow&oq=esp32+wificlient)), quindi se più richieste arrivano mentre l'ESP è impegnato nell'upload, dovrebbe essere "valida" solo l'ultima ricevuta e di fatte le altre vanno perse.

Comunque questa sera provo a simulare anche io.

droidprova, 2021-06-03:
Ciao, mi riferisco a questo esempio: [https://github.com/cotestatnt/AsyncTelegram2/tree/main/examples/ESP32/ESP32-CAM](url)
Come tempi tra un messaggio e l'altro mi riferisco di più a 2/3 secondi. Comunque per messaggi persi intendo nuovi messaggi, è chiaro che do il tempo al sistema di inviare la precedente foto. Ecco appena arriva la precedente foto, se reinvii una nuova richiesta molto spesso la perde.

Ciao

cotestatnt, 2021-06-03:
Ok, ora il quadro è più chiaro.
Al momento non ho a disposizione l'ESP32-CAM, appena possibile provo a simulare.

droidprova, 2021-06-03:
Per il momento ho un pò di materiale cablato su una bread board e sto facendo molti test sia hardware che software. A volte passo su altra libreria concorrente in quanto ho difficoltà a rendere il mio progetto basato su ESP32-CAM stabile. In quest'ultimo caso noto più costanza e continuità di funzionamento. Se vuoi posso girarti il mio listato.

Ciao e grazie

cotestatnt, 2021-06-03:
Certo, fai pure.

droidprova, 2021-06-03:
Ma da github si possono mandare PM con allegati? Oppure se vuoi ti allego il file.ino da Arduino? Non mi pare elegante mettere il codice qui con il richiamo ad altre librerie....

cotestatnt, 2021-06-03:
Manda pure a cotestatnt@yahoo.com

droidprova, 2021-06-07:
ciao, ho visto che hai apportato delle modifiche alla libreria, per cui sempre riprovando lo stesso esempio, ottengo lo stesso comportamento:

**Ciao, mi riferisco a questo esempio: https://github.com/cotestatnt/AsyncTelegram2/tree/main/examples/ESP32/ESP32-CAM
Come tempi tra un messaggio e l'altro mi riferisco di più a 2/3 secondi. Comunque per messaggi persi intendo nuovi messaggi, è chiaro che do il tempo al sistema di inviare la precedente foto. Ecco appena arriva la precedente foto, se reinvii una nuova richiesta molto spesso la perde.

Ciao**

il debug è questo:

[E][AsyncTelegram2.cpp:167] getNewMessage():	deserializeJson() failed with code{
  "ok": false,
  "error_code": 400,
  "description": "Bad Request: file must be non-empty"
}

Ciao

cotestatnt, 2021-06-09:
Ciao @droidprova.
In realtà non ho ancora avuto modo di fare delle prove (le modifiche le avevo apportate poco prima) perché sono stato assente per tutto il weekend.
Dal messaggio di errore, sembrerebbe che cerchi di inviare un file "vuoto" o comunque non esistente, quindi credo che l'errore sia a livello di "esempio" e non di libreria in se (o meglio, la libreria in effetti potrebbe verificare se il file è "congruente").

Appena riesco a dedicarci un po' di tempo ti faccio sapere.

droidprova, 2021-06-11:
Ok grazie, attendo tue...
Ciao

cotestatnt, 2021-06-11:
Ciao, come sospettavo l'errore è nella scrittura del file e non nel successivo invio.
Non sono riuscito a capire perché, ma ogni tanto l'istruzione `file.write(fb->buf, fb->len); ` non lavora come ci si aspetta ed il file risultante ha dimensione 0 byte.
Per ora ho risolto controllando subito dopo la dimensione e se == 0 eseguo nuovamente l'istruzione di scrittura (scrittura ed invio a Telegram ora sono in due funzioni distinte).

La soluzione non è di certo la migliore o elegante, ma al momento non ho molto tempo per indagare in modo più approfondito.

EN:
_Hi, as I suspected the error is suring file saving and not in the subsequent sending.
I couldn't figure out why, but sometimes instruction `file.write (fb-> buf, fb-> len); `does not work as expected and the resulting file is 0 bytes in size.
For now I temporaly solved it by checking immediately after the size and if == 0, write instruction is called again (writing file and sending to Telegram are now in two distinct functions).

The solution is certainly not the best or elegant, but at the moment I don't have much time to investigate further._

droidprova, 2021-06-11:
ok grazie mille, lo metto in prova e ti aggiorno nei prossimi gg. Buon weekend!

droidprova, 2021-06-15:
Dunque dopo il tuo ultimo aggiornamento le cose sono migliorate di molto, ma a volte qualche messaggio contenente l'invio di foto continua a perdersi. Invece l'invio/ricezione di messaggi di solo testo vanno una meraviglia.

Ciao

closed 2021-06-23 by cotestatnt