cotestatnt/AsyncTelegram2 #6: IL vecchio ESPTelegramBOT....

migreco2, 2021-06-11:
Ciao Tolentino
Usavo tantissimo il precedente ESPTelegramBOT che non so per quale ragione mi da sempre Message not delivered....mi viene il dubbio che ho aggiornato qualche libreria di IDE senza rendermi conto...
Lo hai eliminato? Forse perché usava ArduinoJson 5?
Mi sai aiutare?
Grazie
Mic

cotestatnt, 2021-06-12:
Perdonami @migreco2, ma ESPTelegramBOT non è una mia libreria. Non saprei come aiutarti, mi spiace.

closed 2021-06-12 by cotestatnt

migreco2, 2021-06-12:
Scusami…. La vedevo collegata a te e poi cancellata
Qui:

https://freesoft.dev/program/109600855

Comunque ho risolto … era la versione della ESP32 dev che deve essere 1.0.4 e la 1.0.6 non era più compatibile
Grazie comunque