cotestatnt/AsyncTelegram2 #7: motion detect & send picture

droidprova, 2021-06-14:
Ciao Tolentino, attravero un PIR il cui pin è gestito tramite interrupt, vorrei che quando viene rilevato un movimento, il sistemi mi invii un messaggio di allert con tanto di foto. Ho scritto questa funzione ma non va:

`void motion_detect() {
TBMessage msg;
  if (flag_auto_detect) {  // modalità motion detect inserita 
    if (motionDetected) {  // rilevo movimento 
      Serial.println("beccato!");
      char* filename = "/frame.jpg";
      if (savePicture(filename)) {
        if (sendPicture(msg, filename))
          Serial.println("Picture sent successfull");
      }
    }
  }
}
`
Quando il PIR rileva il movimento, vedo il flash accendersi ma la foto non viene spedita. Nella funzione che ho scritto ci sono errori? Come faccio al di fuori della funzione di telegram a gestire l'invio di messaggi quando si concretizzano degli eventi? 
Grazie

cotestatnt, 2021-06-14:
Non può funzionare perché la struttura `TBMessage msg;` è vuota e non contiene l'id della chat alla quale spedire il messaggio.
Aggiungi l'istruzione 
`msg.sender.id = 123456789; // metti il tuo id`

droidprova, 2021-06-14:
Grazie 1000! Provo subito...

droidprova, 2021-06-15:
Si può chiudere. Tutto ok come da tue indicazioni.

Grazie!

closed 2021-06-23 by cotestatnt