cotestatnt/AsyncTelegram2 #9: CI / DC

shafr, 2021-06-24:
So i'm trying to add some actions to make life easier:
* Build library
* Build examples
* CppLint

Please share your thoughs on items below:

## Build libarary:
What are the dependencies for Ardunio Uno ? The main issue was with `functional` - is that in `<StandardCplusplus>` library ?

## Examples:
* I need to figure out how to compile multi-file examples 
* ESP32 have some weird issues with `openslab-osu/SSLClient`

##  CPP Lint:
not sure if linting is needed at this point (it fails with 197 issues). If you want, I can try to find some pull-request-review-style that would say which lines have isuses for easier processing.

##  Issues template
For easier troubleshooting (like instruction for stack trace decoding if present), board version, etc

## Pull request template

6) Probably also sketch size comparison on pull request.
https://github.com/arduino/report-size-deltas

cotestatnt, 2021-06-24:
Hello @shafr 
Thanks for your suggestion.

- Arduino Uno, as far as i know can't run a SSL connection, so no support for this board is feasable. The [**functional** C++ ](https://www.cplusplus.com/reference/functional/function/) include, is not avalaible with avr-gcc. This library is necessary only for defining the callback function of inline keyboard buttons
`using CallbackType = std::function<void(const TBMessage &msg)>; `
It's not a big dependency so probably can be possible find another way, but the SSL problem with small AVR mcus remain.
Have you a working examples with Uno connected as client of HTTPS server?

- "Explicit library" what do you mean? Can you explain me more in detail the issue?

- Regarding linting, i will shurely do it before the pull request in order to add library to Arduino Library Manager

- Good idea for templates! I will work on it :)

shafr, 2021-06-24:
1) what arduino library do you plan to use? or itll be only esp-based?
i havent used arduino for ages.

2) explicit means that i cannot use definition with 'repo/library : version', only github repo link. im not sure why.

cotestatnt, 2021-06-28:
No, it's not only esp-base but the MCU must be able to handle a SSL stream and the ATMega328 of Uno board **can't do this job** (as I know).
You have to use the complete github link because I have not yet requested to add the library to the list in the library manager.

shafr, 2021-06-28:
Well that also applies for other libraires, there is some strange error message about fetching libraries, something with `to_lowercase`. 
So I'll leave ESP8266 & ESP32 for now then ?

shafr, 2021-07-09:
Also if you wan't I can copy workflow to show Code / LFS size on pull request, like here:
https://github.com/shafr/esp-canary/pull/21

cotestatnt, 2021-07-11:
Yes, why not.

A question from my side...
I'm having trouble with workflow **compile_examples.yml**. 
It fails on _sendPhoto.ino_ example with esp8266 platform, but the example compile and run with no problem both with Arduino IDE or VSCode+PlatformIO.

For the moment I've commented out that line

shafr, 2021-08-17:
I had forked it and it is compiling ok. Maybe there was some bug before  ?

shafr, 2021-08-17:
Probably you can pin this thread for future requests

shafr, 2021-08-17:
Also I've compared sizes for both AsyncTelegram1 & AsyncTelegram2 with [my project](https://github.com/shafr/esp-canary/pull/25) :

| Board  | \|  | ESP8266 RAM | ESP8266 Flash | \|  | ESP32 RAM    | ESP32 Flash  |
|--------|-----|-------------|---------------|-----|--------------|--------------|
| V1 | \|  | 51.7% | 64.5%   | \|  | 15.4% | 96.3% |
| V2   | \|  | 49.0%   | 64.9%     | \|  | 15.2%   | 96.3%   |

closed 2022-01-23 by cotestatnt