cotestatnt/AsyncTelegram2 #10: welcome message

droidprova, 2021-06-29:
Ciao Tolentino, nei vari esempi nel setup() trovo questo blocco:

`char welcome_msg[128];
  snprintf(welcome_msg, 128, "BOT @%s online\n/help all commands avalaible.", myBot.getBotName());
  int32_t chat_id = 436865110; // You can discover your own chat id, with "Json Dump Bot"
  myBot.sendTo(chat_id, welcome_msg);`

se volessi spedire questo "messaggio di benvenuto" a 2 utenti Telegram, sai come posso fare?

Grazie 1000!

cotestatnt, 2021-06-30:
ciao @droidprova, per inviare lo stesso messaggio a più utenti, il modo più efficace è creare un gruppo al quale aggiungi, oltre agli utenti interessati, anche il tuo bot.

Il bot deve essere abilitato ai gruppi con **BotFather** _(Bot Settings -> Allow Groups -> Yes)_
Una volta fatte queste operazioni preliminari, devi recuperare l'id del gruppo (si tratta di un numero **intero negativo**)

Uno dei modi più semplice per trovare l'id è aggiungere questo ulteriore bot al gruppo, [@getidsbot](https://t.me/getidsbot), per poi rimuoverlo subito dopo. Questo bot ti risponderà facendo il dump di tutte le caratteristiche del gruppo, id compreso.

Se non vuoi usare i gruppi devi recuperare l'id di ciascun utente ed inviare singolarmente lo stesso messaggio, ma francamente te lo sconsiglio. Molto più pratico con i gruppi.
```
  char welcome_msg[128];
  snprintf(welcome_msg, 128, "BOT @%s online\n/help all commands avalaible.", myBot.getBotName());
  // Send a message to a Telegram group (negative ID)
  int32_t group_id = -123456789;  
  myBot.sendTo(group_id, welcome_msg);
```

English:
In order to send the same message to multiple users, the most effective way is to create a group to which you add your bot as well as interested users.

The bot must be enabled for groups with **BotFather** _(Bot Settings -> Allow Groups -> Yes)_
Once these preliminary operations have been done, you have to find the group id (this is a **negative integer**)

One of the easiest way is add this other bot to the group as well (and then remove it immediately afterwards) [@getidsbot](https://t.me/getidsbot). This bot will reply dumping all the group infos.

droidprova, 2021-06-30:
Grazie Tolentino, diciamo che nell'applicazione che sto utilizzando, a colloquiare con telegram, sono solo 2 utenti che tra l'altro inseriscono i loro ID in una maschera di configurazione basata su wi-fi manager, quindi reperire gli ID non è un problema e il basarsi su un ID di gruppo è troppo. 

A questo punto se faccio questo:

`char welcome_msg[128];
  snprintf(welcome_msg, 128, "La tua board è online.\n/start per abilitare il pannello di controllo", myBot.getBotName());
  int32_t chat_id = EEPROM.get(addr + 293, userID1); 
  myBot.sendTo(chat_id, welcome_msg);
  int32_t chat_id = EEPROM.get(addr + 297, userID2); 
  myBot.sendTo(chat_id, welcome_msg);`

ottengo questo errore: `redeclaration of 'int32_t chat_id'`  come mi aspettavo, dove appoggio il  2° ID?

Grazie 1000!

cotestatnt, 2021-06-30:
L'errore è dovuto al fatto che tenti di dichiarare 2 volte una variabile con lo stesso nome
`int32_t chat_id `
Inoltre` EEPROM.get() `non si usa in quel modo, quindi l'istruzione che assegna a `chat_id` non ha senso.

```
char welcome_msg[128]; 
snprintf(welcome_msg, 128, "La tua board è online.\n/start per abilitare il pannello di controllo", myBot.getBotName()); 
EEPROM.get(addr + 293, userID1); 
myBot.sendTo(userID1, welcome_msg); 
EEPROM.get(addr + 297, userID2); 
myBot.sendTo(userID2, welcome_msg);
```
userID1 e userID2 devono essere due variabili di tipo **int32_t** ovvero **long** 
In realtà ai nuovi utenti ultimamente viene assegnato un id che supera la capacità di memorizzazione di una variabile a 32 bit, quindi se non vuoi incontrare possibili problemi è meglio se dichiari userID1 e userID2 come **int64_t** equivalenti a **long long**

droidprova, 2021-06-30:
Grazie, funziona!
So che gli ultimi ID telegram che ho potuto avere visione sono da 10 caratteri, mi stai dicendo che siamo a 11 adesso?

cotestatnt, 2021-06-30:
https://core.telegram.org/bots/api#chat

_This number may have more than 32 significant bits and some programming languages may have difficulty/silent defects in interpreting it. But it has at most 52 significant bits, so a signed 64-bit integer or double-precision float type are safe for storing this identifier._

closed 2021-07-09 by cotestatnt