cotestatnt/AsyncTelegram2 #16: change parse_mode to "MarkdownV2" or "HTML"

DerDoubleD, 2021-08-06:
how can i change the parse_mode to "MarkdownV2" or "HTML"?

best regards
Daniel

cotestatnt, 2021-08-06:
``..........
  ..........
  // local variable to store telegram message data
  TBMessage msg;

  // if there is an incoming message...
  if (myBot.getNewMessage(msg)) {    
    // Send a message to your public channel
    String message ;
    message += "<b>Bold Text</b>";
    msg.isHTMLenabled = true;
    // msg.isMarkdownEnabled = true;
    myBot.sendMessage(msg, message);
  }
  .........
  .........
```

DerDoubleD, 2021-08-06:
> ```
> ........
>
>   // local variable to store telegram message data
>   TBMessage msg;
>
>   // if there is an incoming message...
>   if (myBot.getNewMessage(msg)) {
>     // Send a message to your public [...]

Thanks a lot, that is exactly what I have been looking for!

closed 2021-08-06 by DerDoubleD