cotestatnt/AsyncTelegram2 #17: Is it possible to set BotCommands?

DerDoubleD, 2021-08-06:
See https://core.telegram.org/bots/api => setMyCommands

Thanks and best regards
Daniel

cotestatnt, 2021-08-06:
I will try to add this functionality as soon as possible.

cotestatnt, 2021-08-10:
Hello @DerDoubleD 
I've just released new version including methods for custom commands.

In order to test, try to add this instructions in your setup:
  ```
// Clear all custom commands
  myBot.deleteMyCommands();

  // Add custom commands to bot (command, description);
  myBot.setMyCommands("help", "Help command");
  myBot.setMyCommands("reply_keyboard", "Reply test keyboard");
  myBot.setMyCommands("inline_keyboard", "Inline test keyboard");

  // Print a JSON serialiazed list of custom commands
  String cmdList;
  myBot.getMyCommands(cmdList);
  Serial.println(cmdList);
```

DerDoubleD, 2021-08-11:
Hi cotestatnt 

i tested it and it works perfect!
Many Thanks for the great Work!!!

Best regards
Daniel

closed 2021-08-11 by DerDoubleD