cotestatnt/AsyncTelegram2 #18: How to interface PIR (HIGH when movement is detected)

FBMinis, 2021-08-10:
Thank you for your new library, the old one was already good but this is even better.
You can answer in italian if you want, I understand it but can't write it very well.

I have some of these PIR in the photos and the output goes HIGH for up to 5min (user regulated) after movement is detected.

Do I just need to change the following lines or is there a better way?

```
// PIR Motion Sensor setup
  pinMode(PIR_PIN, INPUT);
```

and 

```
// PIR motion detected 
    // could be on interrupt, but PIR sensor usually keep signal HIGH for enough time
    if(digitalRead(PIR_PIN) == HIGH) {   
      currentPict = 0;
      waitPhotoTime = 0;
      char message[50];
      time_t now = time(nullptr);
      tInfo = *localtime(&now);
      strftime(message, 50, "%d/%m/%Y %H:%M:%S - Motion detected!", &tInfo);
      Serial.println(message);
      myBot.sendTo(userid, message);
    }
```
[image]
[image]

cotestatnt, 2021-08-10:
Hi @FBMinis.
Thaks, but I think it's better in English for other users that might search for something like you.

Regarding your question, 5 minutes is a very long time and with that code the bot will send a lot of pictures, not needed probably. Maybe a rising edge flag can help you.

```
   static bool risingEdge = false;    // Static (or global) bool var to catch the rising edge of signal
    
   // could be on interrupt, but PIR sensor usually keep signal HIGH for enough time
   if(digitalRead(PIR_PIN) == HIGH  && !risingEdge) {  
      risingEdge = true;
      currentPict = 0;
      waitPhotoTime = 0;
      char message[50];
      time_t now = time(nullptr);
      tInfo = *localtime(&now);
      strftime(message, 50, "%d/%m/%Y %H:%M:%S - Motion detected!", &tInfo);
      Serial.println(message);
      myBot.sendTo(userid, message);
   }
   
   // After a while signal come back to LOW level, so we can clear the bool flag
   if(digitalRead(PIR_PIN) == LOW && risingEdge) {  
     risingEdge = false;
   }
```

cotestatnt, 2021-08-22:
> Maybe GPI013 must be HIGH during boot because of SDCard.

Yes, this can be the reason.

closed 2021-08-23 by cotestatnt