cotestatnt/AsyncTelegram2 #19: USE_CLIENTSSL ESP32

shafr, 2021-08-17:
Hi, When using ESP32 & USE_CLIENTSSL = true, flash memory exceedes 100% and compilation fails.
Can you please explain what does that option means ?

cotestatnt, 2021-08-22:
Hi @shafr, sorry for late reply but I was on holiday.
This option is relevant only with examples and is intended for MCUs where SSL handling is not included in the Arduino framework like for example STM32 or SAMD21.

Anyway, I've just tried with one of examples (lightBot.ino) and with ESP32 (with NodeMCU-32S board). The sketch compile and work as expected.
Could you better explain the problem you have identified?

shafr, 2021-08-23:
Hello, thanks for reply.

Here's example of the code i'm using.
Try to compile it locally with true and false - for some reason it would not fit anymore.

https://github.com/shafr/esp-canary/blob/01677ab7d5c363664145d465e3f3b6814d9ab4f9/src/reporting/telegram.cpp#L10

cotestatnt, 2021-08-23:
I've seen your project and there are really a lot of features!

The strange thing is that trying to compile one of the examples with and without the USE_CLIENTSSL option true, the compiled size is smaller when the SSLClient library is used instead of the WiFiClientSecure included in the framework.

However, from what I have been able to see, with the latest versions of the ESP32 framework for Arduino, they have solved many bugs and the WiFiClientSecure library performs just as well.

If you want to test using SSLClient you can try [changing the partition scheme](https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables).
For example with the pre-configured option
``` 
board_build.partitions = min_spiffs.csv 
``` 
added to the **platformio.ini** file, project compile, but I don't know if the remaining 190kb filesystem space is enough for your project.

shafr, 2021-08-24:
Frankly I was amazed, how does Tasmota, EspHome, MarlinFw are able to do all the functionality with such few memory. I've just recently learned about ESP8266 memory allocation (1mb at most for application code), but haven't fully figured out ESP32 one.

Thanks for the heads up, will check that.

closed 2021-08-31 by cotestatnt