cotestatnt/AsyncTelegram2 #21: Unicode Command Option

garudaonekh, 2021-09-22:
Hello,
How can we use Unicode in command?
When I use Unicode, the command disappear, it shows as normal text.

For example: /\u2010\u2030

cotestatnt, 2021-09-23:
Telegram has only two kind of text parser [(text formatting options)](https://core.telegram.org/bots/api#formatting-options)

- HTML style
- MarkdownV2 style

You could switch to HTML entity equivalent of your special characters.
For example the character **&#8240;** corresponding to unicode **\u2030** is equal to the string"& #8240;" (I have  added an unneeed space otherwise the Github parser show only the character).

For example, this code will reply to every message with this sentence:

_Message with HTML code inside.
 6.58 &#8240;_ 

```
void loop() {  
  // local variable to store telegram message data
  TBMessage msg;

  // if there is an incoming message...
  if (myBot.getNewMessage(msg)) {    
    // Send a message to your public channel
    msg.isHTMLenabled = true;

    String message ;
    message += "Message with HTML code inside.\n 6.58 &#8240; ";

    // echo the received message
    myBot.sendMessage(msg, message);
  }
}
```

[image]

Check here for HTML entity characters of whatever you need
https://www.fileformat.info/info/unicode/char/search.htm

garudaonekh, 2021-09-27:
yes, it work for normal message. But If i want to use command text as unicode, it is not working. 

```
myBot.sendMessage(msg,
            "System:\n/restart for restart MCU\n"
            "/&#8240;dd  for local system timedate\n\nScheduler:\n"
            "/add to add new event to list");
      }
``
<img width="423" alt="Screen Shot 2021-09-27 at 3 34 32 PM" src="https://user-images.githubusercontent.com/2074857/134873638-682c8a3d-daa5-4f7e-a9f1-9a3832fe1500.png">
`