cotestatnt/AsyncTelegram2 #22: Question about chat id

garudaonekh, 2021-09-22:
Hi,
I see that if the bot have 1000 users, so any messages from all users will be sent to all devices.

Is there a way to send only message from a specific chat id to a specific device?

Thanks;

garudaonekh, 2021-09-24:
I need to have several devices sharing the same bot. And devices are bound by chat-id for different users. Bjt this look not possible

cotestatnt, 2021-09-24:
If your devices are all in the same network, you can use a master/slave topology where only the master talk with the Telegram BOT.
Then, from device to device, you can use another type of communication like for example ESP-NOW.

garudaonekh, 2021-09-26:
different network. I am thinking of using a public cloud PHP or Node.Js acting as middleware

cotestatnt, 2021-09-26:
A possible solution for your purposes could be **Google Firebase** which is free, (ore better, within very very large limits), reliable and does not require a specific server.

Check out [this library!](https://github.com/mobizt/Firebase-ESP-Client)

For example, for my own test, I've written a sketch with which you can control the esp gpio remotely with a simple web page (using the personal space offered by Github, but it can be hosted everywhere, even on the ESP flash memory).  

[Sources (HTML, JS, CSS and ESP firmware)](https://github.com/cotestatnt/cotestatnt.github.io)
[Actual online webpage](https://cotestatnt.github.io/)

garudaonekh, 2021-09-27:
What I like about Telegram is it's easy for users to control without installing any new app or going to any website. People are already familiar with Telegram. It's very popular in my country.

cotestatnt, 2021-09-27:
Yes, I can understand your point of view.
Thinking about your requirements, maybe I came up with a possible solution.

Since the only thing that prevents to run with the same token on multiple ESP devices is the polling mechanism towards the Telegram server, you could delegate this task to a single ESP that will act as a gateway for the others through a second communication channel.

On the other devices, you have to avoid calling the **getNewMessage()** method and no conflict will occur.
This method calls in background the **checkConnection()** method wich is responsable to handle connection with Telegram server, so when you need to send a message from one of "non-gateway" device you have to call it manually to establish connection.
But, in order to call manually you have to put this method in the public section of class because actually it is private.

For the message delvery to other devices, firebase, as said, can be a great choice thanks to "[stream callback](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/DataChangesListener/Callback/Callback.ino)" mechanism (you can define a function that run everytime datas will be updated), but also a websocket or a MQTT  server could do just as well.
I've just tested, and it works.

garudaonekh, 2021-09-28:
I plan to use this scenario for master/slave projects for farm. In which, we'll have an ESP32 as main controller and communicate with other slave esp32 devices for each greenhouse in the whole farm. 

However, the main objective of this post is about different ESp32 for different users(not connected with each others). The customer journey I want is: 

a. A user will scan QR on the devices and it'll go to the Telegram bot which already created by me
b. The device is connected to the Internet thru 2g SIM GPRS. I used T-Call Sim 800L.
c. The device then will wait for its owner to register.
d. The user will then key in the unique id number written on the device in the bot chat. The new unregistered device will pick the message and see that the ID number belong to it, will register that user id as its owner.

Thus based on the this scenario, I think it's important to have a middleware software that will stand between telegram and the device. The missing part for me is the communication between the device and the middleware which I don't know what kind of technology stack to use. What do you think?

closed 2022-01-16 by cotestatnt