cotestatnt/AsyncTelegram2 #23: Not an issue, need help sending a photo when powering on

FBMinis, 2021-09-28:
I would like to send a photo to my bot when the esp32cam is powered on, then do nothing else (nothing in the loop function). That is, I would like it to send a photo without needing the /TakePhoto command. Thank you.

FBMinis, 2021-09-29:
I got it working like this: I cleared the loop function and added the following code at the end of the setup function, after init_camera():

 // Take and send photo
  TBMessage msg;
  String text = "/takePhoto";
  msg.text = text;
  msg.sender.id = userid;
  if (msg.text.equalsIgnoreCase("/takePhoto")) {
    Serial.println("\nSending Photo from CAM");
    if (sendPicture(msg))
      Serial.println("Picture sent successfull");
    else
      myBot.sendMessage(msg, "Error, picture not sent.");
  }

cotestatnt, 2021-09-29:
Hi @FBMinis and sorry for late response.

It's not necessary handle the text message, just send it without checking the "/takePhoto" string :

 ```
  TBMessage msg;
  msg.sender.id = userid;

  Serial.println("\nSending Photo from CAM");
  if (sendPicture(msg))
      Serial.println("Picture sent successfull");
  else
      myBot.sendMessage(msg, "Error, picture not sent.");
```

closed 2021-09-29 by cotestatnt

FBMinis, 2021-09-29:
Ok, it's working, thank you.

I found that taking a photo immediately after `init_camera() ;` resulted in over-exposed photos. I guessed the sensor needed time to stabilize, therefore I moved `init_camera() ;` above `WiFi.begin(ssid, pass);` which solved the situation, the photos look good now.
[image]
[image]