cotestatnt/AsyncTelegram2 #27: Send Bold messages

pablopeu, 2021-12-06:
Hi Tolentino, is it possible to send messages using telegram markup? in any chat if you put anything between ** its sent as bold
Also tried these and didn't work: https://core.telegram.org/api/entities
Thanks

cotestatnt, 2021-12-06:
Hi @pablopeu, thanks for using this library!
Regarding your question, both Markdown v2 and HTML are supported from AsyncTelegram2, but by default they are disabled.
I'm considering leaving HTML formatting on by default for the next realease.

You have to take care with Markdown, because Telegram Bot support [only some features ](https://core.telegram.org/bots/api#formatting-options) while HTML is more flexible.

Anyway, if you want send a formatted message, you can do in this way:
```
  TBMessage msg;
  msg.sender.id = 1234567890;   // You can discover your own chat id, with https://t.me/getidsbot
  
  String message = " *Bold text*\n _Italic text_\n __underline__\n ~strikethrough~";
  msg.isMarkdownEnabled = true;
  myBot.sendMessage(msg, message);

  // Or with HTML parser
  String message = " <b>Bold text</b>\n <i>Italic text</i>\n <u>underline</u>\n <s>strikethrough</s>";
  msg.isHTMLenabled = true;
  myBot.sendMessage(msg, message);
```

[image]

pablopeu, 2021-12-06:
One more question, the message: msg.isMarkdownEnabled = true; 
needs to be sent every time or its enable and forget?

Thanks

pablopeu, 2021-12-06:
Hi Tolentino, please help a newbie that barely dabs at arduino flavor of C, how do I dispatch a message adding this config?

I assemble the message string (declared globally) and then call this routine to send the message, this way I can add multiple userid if needed in a single place.

```
void despachar(){
//  myBot.sendTo(userid, message); //Juan
//  userid=888888; //edited
  myBot.sendTo(userid, message); 
  userid=999999; // edited
  message = ""; 
}
```

On a side note, tried it modifiing the declarations file and using markup I had troubles parsing some strings, but then I disabled markup and enabled html and it works like a charm.

cotestatnt, 2021-12-06:
For shure! But please le me understand exactly what you want to achieve.
The function `void despachar()` will send messages always to the same user? Would you like to pass parameters like the body message itself for example?

pablopeu, 2021-12-06:
THANKS!!
yes, the message is sent to the users defined in despachar()

here is another routine where the message is assembled:

`void maxt(){
  String texto=msgText.substring(8);
  maxtemp=texto.toInt();
  message += "<b>CONFIGURACIÓN:</b> La alarma de alta temperatura esta configurada en: ";
  message += maxtemp;
  message += "C";
  message += "\n";
  grabar(); //here I save the data to eeprom
  despachar();  //calls the routine to send the message to the user or users
}`

These routines are called from here:

` TBMessage msg; // hay mensajes de Telegram?
  if (myBot.getNewMessage(msg)) {  
    msgText = msg.text;
    if (msgText.startsWith("/ayuda"))   ayuda();
    if (msgText.startsWith("/frecu"))   frec();
    if (msgText.startsWith("/mintemp")) mint();
    if (msgText.startsWith("/maxtemp")) maxt();
    if (msgText.startsWith("/alarma"))  alarma();
    if (msgText.startsWith("/reporte")) reportar();
    if (msgText.startsWith("/config"))  confi();
  }`

in this last piece of code, when I used markup /reporte and /ayuda didn't work, but with html worked.

cotestatnt, 2021-12-06:
So if I have undesrtood well, the problem is that `sendTo() `don't have support for TBMessage input parameter that you need for the formatting markup options (I will fix also this in the next release).

`sendMessage() ` and `sendTo() ` are basically the same thing except that with sendTo you have to explicit use the ID of user (or the ID of a group).
```
int64_t userid = 99999999;  // Set this globally
String message = "";        // Also this is global

void despachar(){
  TBMessage msg;
  msg.sender.id = userid ; 
  msg.isHTMLenabled = true;
  myBot.sendMessage(msg, message);
  message = ""; 
}
```

But, using local String variable is a better approach in my opinion which is less prone to memory fragmentation
```
int64_t userid = 99999999;  // Set this globally

void maxt(){
  String message = "<b>CONFIGURACIÓN:</b> La alarma de alta temperatura esta configurada en: ";
  message += msgText.substring(8).toInt();
  message += "°C\n";
  grabar(); //here I save the data to eeprom
  despachar(message);  //calls the routine to send the message to the user or users
}

// return true if message is successful delivered
bool despachar(String & theMessage){
  TBMessage msg;
  msg.sender.id = userid;
  msg.isHTMLenabled = true;
  return myBot.sendMessage(msg, theMessage);
}
```

pablopeu, 2021-12-06:
Thanks, will test this ASAP!

PS: I see what you did here 
> message += "°C\n";
LMAO thanks!!

pablopeu, 2021-12-07:
This is still intriguing me:

> sendMessage() and sendTo() are basically the same thing except that with sendTo you have to explicit use the ID of user (or the ID of a group).

This means I can send message to whoever started the bot and there is no need to specify a single userid?

cotestatnt, 2021-12-07:
Yes. If you look at the examples included in library, you can see in the `loop()` function this kind of instructions:
 ```
  // local variable to store telegram message data
  TBMessage msg;
  // if there is an incoming message...
  if (myBot.getNewMessage(msg)) {    
  ......
```
If a new message is available, the library will set TBMessage struct data, passed as parameter to `getNewMessage()`, with all informations about sender: user id, text of message, type of message etc etc.

pablopeu, 2021-12-11:
Hi again Tolentino, I'm having a problem with messages sent as HTML, here is my message:

`void ayuda(){
  String message = "<b>AYUDA:</b> Lista de comandos.";
  message += "\n";
  message += "\n";
  message += "Los comandos en donde aparece <b>xx</b> hay que reemplazar <b>xx</b> por un valor numérico, por ejemplo <b>/mintemp 2</b>";
  message += "\n";
  message += "\n";
  message += "<b>/frecu xx</b> Tiempo entre notificaciones de temperatura (en horas)";
  message += "\n";
  message += "\n";
  message += "<b>/maxtemp xx</b> Alarma de alta temperatura (en grados)";
  message += "\n";
  message += "\n";
  message += "<b>/mintemp xx</b> Alarma de baja temperatura (en grados)";
  message += "\n";
  message += "\n";
  message += "<b>/alarma  xx</b> Frecuencia de envio de avisos de alarma (en minutos)";
  message += "\n";
  message += "\n";   
  message += "<b>/reporte</b> Lectura de los valores actuales";
  message += "\n";  
  message += "\n";  
  message += "<b>/config</b> Muestra la configuración actual";
  message += "\n";
  message += "\n";
  message += "<b>/ayuda</b> Envía esta lista de comandos";
  message += "\n";

  despachar(message);
 
}
// return true if message is successful delivered
bool despachar(String & theMessage){
  TBMessage msg;
  msg.sender.id = userid;
  msg.isHTMLenabled = true;
  return myBot.sendMessage(msg, theMessage);
}
`

when telegrams receives it it shows like this:

[image]

but if I send the request to get the same message again the program crashes, I suspect its something in the codes I created which start with a / the same as telegram commands

if I replace all the slashes with the minus character I can send any number of times the request for the message (via the /ayuda command I created) it works flawlessly.

[image]

Does this make any sense?

Also, I noticed something else, but I'm not 100% sure yet, Im testing this arduino program in esp32 and esp8266, when the described behavior happens in esp8266 the program crashes beyond recovery but in esp32 it stops responding until the program generates an automatic report and sends it via a message, then I can send the /ayuda command , and again, it stops responding until the next generated report is sent.

pablopeu, 2021-12-11:
Here is test code to reproduce the crash, im testing it with esp8266:

```
#include <AsyncTelegram2.h>
#include <time.h>

// Timezone definition
#define MYTZ "UTC+3" //timezone https://sites.google.com/a/usapiens.com/opnode/time-zones

#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#ifdef ESP8266
  BearSSL::WiFiClientSecure client;
  BearSSL::Session   session;
  BearSSL::X509List  certificate(telegram_cert);
  
#elif defined(ESP32)
  #include <WiFi.h>
  #include <WiFiClient.h>
  #include <WiFiClientSecure.h>
  WiFiClientSecure client;    

#endif

AsyncTelegram2 myBot(client);
// Telegram BOT Token (Get from Botfather)
const char* token =  "edited";  // Telegram token
//String message="" ; //mensaje armado con el reporte
#define nombrecamara "testcode" //para OTA y Wi-Fi
String msgText="";

// Use @myidbot (IDBot) to find out the chat ID of an individual or a group
// Also note that you need to click "start" on a bot before it can
// message you
int64_t userid = 999999999; // sin poner ID de grupo o de usuario no funciona

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Arrancando...");
  delay(50);
  String message=""; //variable local

// inicio de configuracion wifi
  WiFi.mode(WIFI_STA);
  WiFi.hostname(nombrecamara);
  WiFiManager wm;             //WiFiManager local initialization
  wm.setTimeout(60);
  wm.autoConnect(nombrecamara);
// fin configuracion wifi

#ifdef ESP8266
  // Sync time with NTP, to check properly Telegram certificate
  configTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
  time_t now = time(nullptr);
  //Set certficate, session and some other base client properies
  client.setSession(&session);
  client.setTrustAnchors(&certificate);
  client.setBufferSizes(1024, 1024);
#elif defined(ESP32)
  // Sync time with NTP
  configTzTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
  #if USE_CLIENTSSL == false
    client.setCACert(telegram_cert);
  #endif
#endif

  // Set the Telegram bot properies
  myBot.setUpdateTime(2000);
  myBot.setTelegramToken(token);
 
  // Check if all things are ok
  Serial.print("\nTest Telegram connection... ");
  myBot.begin() ? Serial.println("OK") : Serial.println("NOK");

  Serial.print("Bot name: @");
  Serial.println(myBot.getBotName());

  message="<b>ARRANQUE MONITOR: </b>";
  message+=nombrecamara;
  message += "\n";
  message="<b>AYUDA:</b> Ingresa /ayuda para obtener la lista de comandos";
  message += "\n";
  despachar(message);
  Serial.print("ARRANQUE MONITOR: ");
  Serial.println(nombrecamara);
  
}

void loop() { 
  TBMessage msg; // hay mensajes de Telegram?
  if (myBot.getNewMessage(msg)) {  
    msgText = msg.text;
    Serial.print("texto: ");
    Serial.println(msgText);
    if (msgText.startsWith("/ayuda"))   ayuda();
    msgText ="";
  }
  
  
}  // FIN DEL PROGRAMA

// return true if message is successful delivered
bool despachar(String & theMessage){
  TBMessage msg;
  msg.sender.id = userid;
  msg.isHTMLenabled = true;
  return myBot.sendMessage(msg, theMessage);
}

//// minus character version
//void ayuda(){
//  String message = "<b>AYUDA:</b> Lista de comandos.";
//  message += "\n";
//  message += "\n";
//  message += "Los comandos en donde aparece <b>xx</b> hay que reemplazar <b>xx</b> por un valor numérico, por ejemplo <b>-mintemp 2</b>";
//  message += "\n";
//  message += "\n";
//  message += "<b>-frecu xx</b> Tiempo entre notificaciones de temperatura (en horas)";
//  message += "\n";
//  message += "\n";
//  message += "<b>-maxtemp xx</b> Alarma de alta temperatura (en grados)";
//  message += "\n";
//  message += "\n";
//  message += "<b>-mintemp xx</b> Alarma de baja temperatura (en grados)";
//  message += "\n";
//  message += "\n";
//  message += "<b>-alarma  xx</b> Frecuencia de envio de avisos de alarma (en minutos)";
//  message += "\n";
//  message += "\n";   
//  message += "<b>-reporte</b> Lectura de los valores actuales";
//  message += "\n";  
//  message += "\n";  
//  message += "<b>-config</b> Muestra la configuración actual";
//  message += "\n";
//  message += "\n";
//  message += "<b>-ayuda</b> Envía esta lista de comandos";
//  message += "\n";
//  despachar(message);
//}

// slash character version
void ayuda(){
  String message = "<b>AYUDA:</b> Lista de comandos.";
  message += "\n";
  message += "\n";
  message += "Los comandos en donde aparece <b>xx</b> hay que reemplazar <b>xx</b> por un valor numérico, por ejemplo <b>/mintemp 2</b>";
  message += "\n";
  message += "\n";
  message += "<b>/frecu xx</b> Tiempo entre notificaciones de temperatura (en horas)";
  message += "\n";
  message += "\n";
  message += "<b>/maxtemp xx</b> Alarma de alta temperatura (en grados)";
  message += "\n";
  message += "\n";
  message += "<b>/mintemp xx</b> Alarma de baja temperatura (en grados)";
  message += "\n";
  message += "\n";
  message += "<b>/alarma  xx</b> Frecuencia de envio de avisos de alarma (en minutos)";
  message += "\n";
  message += "\n";   
  message += "<b>/reporte</b> Lectura de los valores actuales";
  message += "\n";  
  message += "\n";  
  message += "<b>/config</b> Muestra la configuración actual";
  message += "\n";
  message += "\n";
  message += "<b>/ayuda</b> Envía esta lista de comandos";
  message += "\n";

  despachar(message); 
}

```

pablopeu, 2021-12-11:
The exact same code on ESP32 does not crash.

cotestatnt, 2021-12-12:
Hi! @pablopeu. 
I'm trying to reproduce the issue with my setup, but it works always as expected.
Wich version of Arduino Core for ESP8266 are you using now? I'm on the latest avalaible for both ESP8266 and ESP32.

pablopeu, 2021-12-12:
Did you tried the code with slash commands and it works on esp8266 after many /ayuda requests? mine crashes on the 2nd request.
Suspecting about that, the other day I completely deleted de Arduino IDE and its included libraries and reinstalled from scratch from freshly downloaded versions, I'm om latest cores for esp32 and esp8266

In the meantime I disabled HTML and everything works as expected on both cores with the slashes, but this still puzzles me, I'm sure its something about the displayed commands, if you see the screen captures I posted earlier, the commands are highlighted in blue, I didn't do that, when I remove the slash and replace it with a minus sign everything works ok again.

[image]

pablopeu, 2021-12-12:
This puzzles me, I thought it crashed, but 7 minutes later out of the blue it recovered by itself and replied without any intervention on my part, so something is looping or something... lol, I'm lost here. And yes, my wifi is working because I was looking some YouTube videos the whole time...

[image]

cotestatnt, 2021-12-13:
Finally I was able to reproduce this issue, and yes **It's very puzzling**!

It's seem related to the quantity of HTML tag code present in the body of message.
With just a some of them, I can send message without problems, but with more than 5 the sketch will be freezed.

I'm still trying to figure out why: it's a non-sense because it's just a sequence of char and it's passed as is to the WiFi client.
I'm facing more or less the same issue even with markdown formatting option.
With markdown it's necessary to use the escape on some characters not allowed by Telegram API:
<i><b>_ * [ ] ( ) ~ ` > # + - = | { } . !</b></i> must be escaped with the preceding character <b>\\\\</b>

This code don't freeze:
 I've used the `F( )` macro in order to save SRAMS during my tests, but issue seems not related to memory consumption
  ```
  String message = F(
    "*AYUDA:* Lista de comandos\\.\n\n"
    "Los comandos en donde aparece *xx* hay que reemplazar *xx* "
    "por un valor numérico, por ejemplo /mintemp *2*\n\n"
    "/frecu *xx* Tiempo entre notificaciones de temperatura \\(en horas\\)\n\n"
    "/maxtemp *xx* Alarma de alta temperatura \\(en grados\\)\n\n"
    "/mintemp *xx* Alarma de baja temperatura \\(en grados\\)\n\n"
    "/alarma *xx* Frecuencia de envio de avisos de alarma \\(en minutos\\)\n\n"
    "/reporte Lectura de los valores actuales\n\n"
    "/ayuda Envía esta lista de comandos"
  );
```

Written in this way, will freeze:
```
  String message = F(
    "*AYUDA:* Lista de comandos\\.\n\n"
    "Los comandos en donde aparece *xx* hay que reemplazar *xx* "
    "por un valor numérico, por ejemplo */mintemp 2*\n\n"
    "*/frecu xx* Tiempo entre notificaciones de temperatura \\(en horas\\)\n\n"
    "*/maxtemp xx* Alarma de alta temperatura \\(en grados\\)\n\n"
    "*/mintemp xx* Alarma de baja temperatura \\(en grados\\)\n\n"
    "*/alarma xx* Frecuencia de envio de avisos de alarma \\(en minutos\\)\n\n"
    "*/reporte* Lectura de los valores actuales\n\n"
    "*/ayuda* Envía esta lista de comandos"
  );
```

pablopeu, 2021-12-13:
Hi Tolentino, thanks for proving I'm not mad, this drives me nuts. I wasn't sure what it was in the beginning so I started adding libraries one by one suspecting an incompatibility but not, that wasn't it. On ESP32 it works fine for you too?
In the meantime I removed html/markdown altogether because I needed a long term test on my project and this bug was preventing it.
But I do have a test bed to try to find a solution, because I'm sure it will be haunting me again sooner or later...

cotestatnt, 2021-12-14:
Hi Pablo.
Finally I've got the bug!!!

When you send a message to Telegram server, after a while you will receive back a message.
In this reply you have also the message text sent before and a list of all "[entities](https://core.telegram.org/api/entities)" (bold, italic, link etc etc)

Also this message of course is processed, but ArduinoJSON, a library used as dependency, fail to parse this JSON reply. Unfortunaly this error was not recognized until now in AsyncTelegram 2 library and this causes sketch freeze.
Now I've managed this kind of situation checking the deserialize error before parsing content (as done several times in library, but I've missed this specific point of source :@ )

I will write also a test sketch limited to the parsing error in order to open an issue in the [ArduinoJSON repository](https://github.com/bblanchon/ArduinoJson).

pablopeu, 2021-12-14:
Great! So I guess it works on ESP32 due to its bigger memory right?
Please let me know how I can help in testing, I do not know how to code in native C and even less JSON :) :) :)

cotestatnt, 2021-12-20:
Yes it is.
Unluckily ther is no way to shrink the reply from Telegram Server :(

@pablopeu, if you agree I'm going to close this issue.

pablopeu, 2021-12-20:
If there are no solutions, and nothing can be done, then its not a problem... 
Thanks a Ton Tolentino!

closed 2021-12-20 by pablopeu