cotestatnt/AsyncTelegram2 #28: Please, add OTA example for Ethernet platform

woodlist, 2021-12-07:
It is so needed.

cotestatnt, 2021-12-09:
Wich kind of board are you planning to use? OTA is strictly related to the type of MCU.
In the examples, there is one with ethernet. I've tested with a couple of MCU (STM32 and nRF52832/40) and actually i'm planning to test also with RP2040 and ATSAMD21 (Arduino Zero).

You have to keep in mind that with Telegram server, the SSL handshake in mandatory so the board must have power enought.
Uno, Mega etc etc, are boards not able to run SSL communications as far as I know.

woodlist, 2021-12-09:
Thanks for reply. Im planning to use on ESP32.

cotestatnt, 2021-12-09:
That's sound good, because you have all computational power needed.
I've tried to take a look at the libraries for OTA update with ESP32 platform: unluckily  **ESP32 HTTPUpdate library** is meant to be used only with `WiFiClient `or `WiFiClientSecure `class.

In any case, this forcing does not arise from **technical needs**. 
In my opinion the developers have assumed that if there is WiFi no one needs to use an Ethernet connection, so I have a good news for you ;) 
Just redefining methods of `HTTPUpdate `and `HTTPClient `libraries (ad some others small things), you can use `HTTPUpdate `even with EthernetClient througth `SSLClient `and IT WORKS like a charm!

Here you can find a working example (the same as for ESP8266, but adapted and tested for a NodeMCU-32S board and W5500 Ethernet shield). I've uploaded in dev branch, because is a very specific requirement. I will see in the future if is better to load also this example in main branch.

[ESP32 Ethernet OTA example](https://github.com/cotestatnt/AsyncTelegram2/tree/dev/test/ESP32_OTA_password)

P.S.
Compiling the SSLClient library with Arduino IDE takes a very long time. 
I recommend you to use VS Code which has a more effective caching mechanism.

woodlist, 2021-12-12:
> That's sound good, because you have all computational power needed. I've tried to take a look at the libraries for OTA update with ESP32 platform: unluckily **ESP32 HTTPUpdate library** is meant to be used only with `WiFiClient `or `WiFiClientSecure `class.
> 
> In any case, this forcing does not arise from **technical needs**. In my opinion the developers have assumed that if there is WiFi no one needs to use an Ethernet connection, so @woodlist I have a good news for you ;) Just redefining methods of `HTTPUpdate `and `HTTPClient `libraries (ad some others small things), you can use `HTTPUpdate `even with EthernetClient througth `SSLClient `and IT WORKS like a charm!
> 
> Here you can find a working example (the same as for ESP8266, but adapted and tested for a NodeMCU-32S board and W5500 Ethernet shield). I've uploaded in dev branch, because is a very specific requirement. I will see in the future if is better to load also this example in main branch.
> 
> [ESP32 Ethernet OTA example](https://github.com/cotestatnt/AsyncTelegram2/tree/dev/test/ESP32_OTA_password)
> 
> P.S. Compiling the SSLClient library with Arduino IDE takes a very long time. I recommend you to use VS Code which has a more effective caching mechanism.

You're so honest. Thanks so much, dear!

closed 2021-12-12 by woodlist

woodlist, 2021-12-19:
Finally I got a time for testing the sketch. It has been accommodated for GSM network usage. The Telegram connection test was passed successfully (local terminal report). The "welcome" interactive message reaches to the bot. But the actual update always falls, being treated as "File is unavailable. Maybe size limit 20MB was reached or file deleted".

woodlist, 2022-01-25:
> That's sound good, because you have all computational power needed. I've tried to take a look at the libraries for OTA update with ESP32 platform: unluckily **ESP32 HTTPUpdate library** is meant to [...]

@cotestatnt Please, revert back the resource: [https://github.com/cotestatnt/AsyncTelegram2/tree/dev/test/ESP32_OTA_password](url)
It is unavailable now.