cotestatnt/AsyncTelegram2 #30: In connection with closed #28 issue

woodlist, 2021-12-19:
Finally I got a time for testing the sketch. It has been accommodated for GSM network usage. The Telegram connection test was passed successfully (local terminal report). The "welcome" interactive message reaches to the bot. But the actual update always falls, being treated as "File is unavailable. Maybe size limit 20MB was reached or file deleted".

[https://github.com/cotestatnt/AsyncTelegram2/issues/28](url)

woodlist, 2021-12-19:
I have inquired also the "msg.document.file_name" It is also empty.

cotestatnt, 2021-12-20:
Thanks @woodlist for your precious feedback.
I'm going to set-up a GSM test environment in order to check what's happening.

I will inform you about result of my tests.

cotestatnt, 2021-12-20:
Hi @woodlist. I've hot a SIM card and a SIM800C module.

Could you share the test sketch you are trying up now?

woodlist, 2021-12-21:
Dear @cotestatnt, please review the code.
`#include <time.h>
#include "HTTPUpdate.h"
#include <AsyncTelegram2.h>
#include <SSLClient.h> // See the source at https://github.com/OPEnSLab-OSU/SSLClient
#include "trust_anchors.h" // Certifiacte generation utility: https://openslab-osu.github.io/bearssl-certificate-utility/

const char* token =  "xxxxxxxx:xxxxxxxxxxx-xxxxxxxxxxxxxxxxxxx";
int64_t chat_id = xxxxxxxxx;

const char* firmware_version = __TIME__;
const char* fw_password = "update";

// Select your modem:
#define TINY_GSM_MODEM_SIM800
#define TINY_GSM_RX_BUFFER   1024
#define rand_pin             39

#define SerialMon Serial
#define SerialAT Serial2

// Set serial for AT commands
//#define DUMP_AT_COMMANDS
//#define TINY_GSM_DEBUG SerialMon

// TTGO T-Call pins
#define MODEM_RST            5
#define LED_BUILTIN 2

const char apn[] = "internet"; // 
const char gprsUser[] = "";
const char gprsPass[] = "";
const char GSM_PIN[]   = "1111";

#include <TinyGsmClient.h>
TinyGsm modem(SerialAT);                          // sets GSM modem to serial1
TinyGsmClient GSMclient(modem);                   // client for GSM
SSLClient sslClient(GSMclient, TAs, (size_t)TAs_NUM, rand_pin);
AsyncTelegram2 myBot(sslClient);

// https://www.geeksforgeeks.org/find-number-of-days-between-two-given-dates/
using namespace std;
unsigned long Seconds;
// A date has day 'd', month 'm' and year 'y'
struct Date {
  int d, m, y;
};

// To store number of days in
// all months from January to Dec.
const int monthDays[12]
  = { 31, 28, 31, 30, 31, 30,
      31, 31, 30, 31, 30, 31
    };
int year3 = 0;
int month3 = 0;
int day3 = 0;
int hour3 = 0;
int min3 = 0;
int sec3 = 0;
float timezone = 0;

void getconnectedtoGSM() {
  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");

  modem.restart();
  // modem.init();
  String modemInfo = modem.getModemInfo();
[... 226 lines cut ...]
      report += httpUpdate.getLastErrorString();
      myBot.sendMessage(msg, report, "");
      break;

    case HTTP_UPDATE_NO_UPDATES:
      myBot.sendMessage(msg, "HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      myBot.begin();
      myBot.sendMessage(msg, "UPDATE OK.\nRestarting in few seconds...");

      // Wait until bot synced with telegram to prevent cyclic reboot
      while (!myBot.noNewMessage()) {
        Serial.print(".");
        delay(50);
      }
      ESP.restart();

      break;
    default:
      break;
  }

}`

woodlist, 2021-12-23:
Hello @cotestatnt !
I did clean install of Win10 and newest Arduino IDE 1.8.19 Upon introduced above sketch compilation I got some warnings, connected with your library, namely:
`In file included from C:\Users\user\Documents\Arduino\libraries\AsyncTelegram2\src/AsyncTelegram2.h:40,
                 from C:\Users\user\Documents\Arduino\ESP\GPS-Vahe_CTBot_OTA\GPS-Vahe_CTBot_OTA.ino:3:
C:\Users\user\Documents\Arduino\libraries\AsyncTelegram2\src/serial_log.h:10: warning: "__FILENAME__" redefined
 #define __FILENAME__ (strrchr(__FILE__, '\\') ? strrchr(__FILE__, '\\') + 1 : __FILE__)
 
In file included from c:\users\user\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\sys-include\sys\reent.h:503,
                 from C:\Users\user\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\user\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\sys-include\stdio.h:60,
                 from C:\Users\user\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2\cores\esp32/Arduino.h:27,
                 from sketch\GPS-Vahe_CTBot_OTA.ino.cpp:1:
C:\Users\user\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/newlib/platform_include/assert.h:38: note: this is the location of the previous definition
 #define __FILENAME__ (__builtin_strrchr( "/" __FILE__, '/') + 1)`
Seems the Arduino developer team have done a hard work over new release.
I got also "#define LED_BUILTIN 2" redefining warning and have fixed that by commenting out this definition.

woodlist, 2022-01-16:
I would like to share an report. Upon recent lib update compilation above posted warnings gone. Unfortunately right now I'm unable to test the code on real hardware, thus I keep the status of this issue yet open.

cotestatnt, 2022-01-23:
Thanks @woodlist.
Unfortunately I have not been able to set up a test project that uses a GSM modem because at the moment I don't have the necessary hardware.

Have you made any progress?

woodlist, 2022-01-25:
Dear @cotestatnt, greetings.
I do report that the msg.document.file_path  is yet empty on terminal view. Accordingly, the OTA does not work.

woodlist, 2022-01-25:
@cotestatnt Wow, I have fixed the problem. In AsyncTelegram2.cpp
`    doc.file_path = "https://api.telegram.org/file/bot" ;
    doc.file_path += m_token;
    doc.file_path += "/";`
were chaged to 
`    doc.file_path = F("https://api.telegram.org/file/bot") ;
    doc.file_path += m_token;
    doc.file_path += F("/");`
aaaand, it works now!

closed 2022-01-25 by woodlist

woodlist, 2022-01-25:
[https://github.com/cotestatnt/AsyncTelegram2/tree/dev/test/ESP32_OTA_password](url)
@cotestatnt Please, revert back.

cotestatnt, 2022-01-26:
Sorry @woodlist , I was tidying up the repos and forgot to restore the dev branch. Now it's online again.

> Wow, I have fixed the problem. In AsyncTelegram2.cpp...

This is a great news! 
It would be really really appreciated if you want share the code from your working example and create a pull request with your fix in AsyncTelegram2.cpp

Thanks!

woodlist, 2022-01-26:
Dear @cotestatnt 
Having fixed things with AsyncTelegram2 however I was unhappy with overall performance, due to used SSL wrapper [https://github.com/OPEnSLab-OSU/SSLClient](url). Your simple non-blocking LED indicator is an very handy instrument on microcontroller's good standing visual checking. Thus it has been discovered that the MC getting stuck too much frequently, drops the network connection and restarts. Having done a hard work I find another library, available at [https://github.com/govorox/SSLClient](url) which works too much better then used library before. The only headache was the valid telegram certificate obtaining. I have tried to export the cert from MS Edge and Chrome browsers, but those were not the same in first, and the both did not worked in second. Then I have used ready made one at [https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/TelegramCertificate.h](url) and now the performance is really good. As you are suggesting, I will prepare an sketch, which uses the GSM network connection, for your Examples.

woodlist, 2022-02-08:
Having used the SSL wrapper from [https://github.com/govorox/SSLClient](url), seems the certificate, available at [https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/TelegramCertificate.h](url) does not rely to [telegram.org](url), but to Go Daddy Root Certificate Authority. I cannot understand yet, how the certificate of root authority acts as valid one for Telegram. Meanwhile, this certificate is valid until 2037 year. Who can advice, am I wrong? Any attempt to download nor export the Telegram certificate from original server and further use in microcontroller code returned no connection with Telegram server.