cotestatnt/AsyncTelegram2 #31: Info version library

droidprova, 2021-12-28:
Ciao Tolentino, ben ritrovato e auguri di buon Natale e anno nuovo!

Volevo dirti che l'IDE arduino non notifica l'aggiornamento della tua libreria alla versione 2.0.5 e che quindi l'update va fatto manualmente. Effettuando l'aggiormento alla versione 2.0.5 e scaricando lo zip "AsyncTelegram2-2.0.5.zip" arduino però mi notifica che la versione installata è ancora la 2.0.4. Come mai?
Inoltre nel file "library.properties" leggo 2.0.4.
So cercando di usare la formattazione bold nel testo di risposta di un messaggio, ma sia in html che markdown v2, i messaggi così formattati non vengono inviati.
Grazie e ciao 

Hello Tolentino, welcome back and best wishes for a Merry Christmas and New Year!

I wanted to tell you that the arduino IDE does not notify the update of your library to version 2.0.5 and that therefore the update must be done manually. By upgrading to version 2.0.5 and downloading the zip "AsyncTelegram2-2.0.5.zip" arduino, however, notifies me that the installed version is still 2.0.4. How come?
Also in the "library.properties" file I read 2.0.4.
I know trying to use bold formatting in the reply text of a message, but in both html and markdown v2, messages so formatted are not sent.
Thanks and bye

cotestatnt, 2021-12-29:
Tanti auguri anche a te!
Grazie per la segnalazione, avevo dimenticato di aggiornare il file _library.properties_
Ho un altro paio di correzioni in cantiere e poi faccio una nuova release.

Per quanto riguarda la formattazione, prova a dare uno sguardo a [questa issue già segnalata e chiusa](https://github.com/cotestatnt/AsyncTelegram2/issues/27).

Let me know!

droidprova, 2021-12-30:
Grazie 1000!

closed 2022-01-03 by cotestatnt