cotestatnt/AsyncTelegram2 #32: Send message to group with esp8266

joroMaser, 2022-01-15:
I have esp8266 and I want to understand how to use this library.

1) How can I listen and send message to telegram group  by group id ? I saw that example https://github.com/cotestatnt/AsyncTelegram2/blob/main/examples/echoBot/echoBot.ino and that send message  to channel , not group

2)How can I send message with inline keyboard like :
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/UniversalTelegramBot.cpp#L607

3)What is the difference between that library and https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot in esp8266 please>

 
Thank you!

cotestatnt, 2022-01-16:
Hi @joroMaser 
Thank for interesting in this library.

 1) In order to send messages to group, first you have to discover the id number (e.g. adding to group @getidsbot bot).
Then you can simply send the message using the method `sendTo()`
 ```
  int64_t groupid = -123456789;
  myBot.sendTo(groupid, "Your message to Group");
```
 2)  But you are using also inline keyboard, so maybe the simplest way for you is to change the recipient id both for user and for chat, once you got new message.  In this way, every messages the bot receive (from users or directly from group), it will reply always to group chat.
The only thing you have to take into consideration is that Telegram adds the username to the text of the messages containing a command. For example **/start** will become **/start@username** so you have to handle in your skecth properly.
 ```
 if (myBot.getNewMessage(msg)) {
    // check what kind of message I received
    MessageType msgType = msg.messageType;
    String msgText = msg.text;
    msg.chatId = groupid;
    msg.sender.id = groupid;
    ......
```
3) (&5) The main difference from this library and others (included Universal-Arduino-Telegram-Bot), is that the polling mechanism with Telegram server is non-blocking. 
In fact, the operating principle is:  connection to the server-> request for new messages->wait reply->response processing.
In AsyncTelegram2 all this steps are executed asynchronously, so while we are waiting for reply from server, MCU can continue with others jobs to be done.
In the examples included, there is always the blinking of a led and as you can see the frequency of the blinking always remains quite constant, which does not happen if you do the same thing with other libraries.
With ESP8266 off course while you are receiving data from server, MCU is engaged in the task and can not do anything else. Depending on the size of the message, this takes some time to download (a few tens of milliseconds).
So if you need a completely asynchronous behavior also in this case, you have to switch to an ESP32 and run everything related to Telegram in a dedicated task.

4) It is certainly possible to do this, but at the moment the library does not provide the necessary tools. I could add a method that allows you to retrieve the entire JSON sent by the server and at that point on the sketch side you can retrieve all the necessary information. I think about it a bit and I'll update you.

joroMaser, 2022-01-16:
Thank you so much about your comment , 
About 2) I want to present for user options for click like:
https://core.telegram.org/file/811140659/1/RRJyulbtLBY/ea6163411c7eb4f4dc

In Universal-Arduino-Telegram-Bot I do:
```
String keyboardJson = "[[{ \"text\" : \" press here please \", \"callback_data\" : \"press\" }]]";
   bot.sendMessageWithInlineKeyboard(chat_id, "Welcome!", "", keyboardJson);
```

And there when user click on bottom I can check by 
`if (bot.messages[i].type == "callback_query")`
I looking for way to do that with your library please

About 5) Do you say that if I used esp8266 and Universal-Arduino-Telegram-Bot the time that I wait for response is blocking and in your library is not blocking ? that is the most important time that I think my esp8266 got reset by software watchdog

cotestatnt, 2022-01-16:
With AsyncTelegram2 you can send inline keyboards (or reply keyboards) both passing directly the JSON string or using the helper methods `addButton() and addRow()` to "build" the keyboards at runtime.

Check this example for how-to in details:
[keyboards.ino](https://github.com/cotestatnt/AsyncTelegram2/blob/main/examples/keyboards/keyboards.ino)
[keyboardCallback.ino](https://github.com/cotestatnt/AsyncTelegram2/blob/main/examples/keyboardCallback/keyboardCallback.ino)

If you want use the String (but would be better with PROGMEM const char* type ), you need to edit a little adding the **inline_keyboard** tag.
```
String keyboardJson = "{\"inline_keyboard\": [[{ \"text\" : \" press here please \", \"callback_data\" : \"press\" }]]}";
myBot.sendMessage(msg, "This is inline keyboard:", keyboardJson);
```

Regarding watchdog reset with UATB, probably you get [here](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/src/UniversalTelegramBot.cpp#L107)

Try to add a **yield();** instruction wich will feed the watchdog in the while loop.

joroMaser, 2022-01-16:
Ok  I have tried to compile this code https://github.com/cotestatnt/AsyncTelegram2/blob/main/examples/ESP8266/OTA_password/OTA_password.ino
But  I got an error :
```

C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.cpp:5:1: error: no declaration matches 'AsyncTelegram2::AsyncTelegram2(Client&)'
    5 | AsyncTelegram2::AsyncTelegram2(Client &client)
      | ^~~~~~~~~~~~~~
In file included from C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.cpp:1:
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.h:85:7: note: candidates are: 'AsyncTelegram2::AsyncTelegram2(const AsyncTelegram2&)'
   85 | class AsyncTelegram2
      |       ^~~~~~~~~~~~~~
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.h:90:5: note:                 'AsyncTelegram2::AsyncTelegram2(Client&, uint32_t)'
   90 |     AsyncTelegram2(Client &client, uint32_t bufferSize = BUFFER_BIG);
      |     ^~~~~~~~~~~~~~
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.h:85:7: note: 'class AsyncTelegram2' defined here
   85 | class AsyncTelegram2
      |       ^~~~~~~~~~~~~~
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.cpp:62:6: error: no declaration matches 'bool AsyncTelegram2::sendCommand(const char* const&, const char*, bool)'
   62 | bool AsyncTelegram2::sendCommand(const char* const &command, const char* payload, bool blocking )
      |      ^~~~~~~~~~~~~~
In file included from C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.cpp:1:
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.h:384:10: note: candidate is: 'bool AsyncTelegram2::sendCommand(const char*, const char*, bool)'
  384 |     bool sendCommand(const char* command, const char* payload, bool blocking = false);
      |          ^~~~~~~~~~~
In file included from C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.cpp:1:
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.h:85:7: note: 'class AsyncTelegram2' defined here
   85 | class AsyncTelegram2
      |       ^~~~~~~~~~~~~~
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.cpp:344:6: error: no declaration matches 'bool AsyncTelegram2::forwardMessage(const TBMessage&, int32_t)'
  344 | bool AsyncTelegram2::forwardMessage(const TBMessage &msg, const int32_t to_chatid)
      |      ^~~~~~~~~~~~~~
In file included from C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.cpp:1:
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.h:157:10: note: candidate is: 'bool AsyncTelegram2::forwardMessage(const TBMessage&, int64_t)'
  157 |     bool forwardMessage(const TBMessage &msg, const int64_t to_chatid);
      |          ^~~~~~~~~~~~~~
C:\Users\MyUser\Documents\Arduino\libraries\AsyncTelegram2-main\src\AsyncTelegram2.h:85:7: note: 'class AsyncTelegram2' defined here
   85 | class AsyncTelegram2
      |       ^~~~~~~~~~~~~~
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

cotestatnt, 2022-01-16:
Are you trying the example just out of the box? (with your token offcourse).
I've just compiled the example with an ESP8266 without problems and it works as expected.

joroMaser, 2022-01-16:
Yes , I put token and I got some compilations error.

cotestatnt, 2022-01-16:
I've checked again deleting all and reinstall with library manager and it works as before, something is wrong with your installation.
From your error log I can see you install with the zip file, try to do a fresh install with library manager

Please past the complete sketch you are trying to compile.

NB.
(in the latest realease, I forgot to set debug messages off)

joroMaser, 2022-01-16:
Ok now it's work , maybe some caching error.

1)
You told me about sendMessageWithInlineKeyboard to send 
```
String keyboardJson = "{\"inline_keyboard\": [[{ \"text\" : \" press here please \", \"callback_data\" : \"press\" }]]}";
myBot.sendMessage(msg, "This is inline keyboard:", keyboardJson);
```
But in this way I send the message to the user that send to message(in private)  . I want to send this message to group by group id
And 
`myBot.sendTo()`
Can't get Json like that , I don't need the msg object at all.

2)Sometimes I got
```

[D][AsyncTelegram2.cpp:144] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:28] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:34] checkConnection():	Connected using Telegram hostname
Last connection was 296 seconds ago

```
Why is that?
Thank you so much!

cotestatnt, 2022-01-16:
`sendTo()` expect a `const char*` type. You can get from String variables with the method `c_str()` wich return the pointer to String buffer. It's simple and safe. 
Loading [from flash with PROGMEM keyword ](https://github.com/cotestatnt/AsyncTelegram2/blob/main/examples/advanced/multipleTimers/scheduler.h#L119)would be even better because you keep more SRAM free.

`myBot.sendTo(123456789, message, keyboardJson.c_str());`

Regarding the messages it's normal.
As written before, I forgot to disable [debug logging](https://github.com/cotestatnt/AsyncTelegram2/blob/main/src/AsyncTelegram2.h#L20) from library in the last release.
The messages is related to the reconnection to Telegram server when it get's missing (after some hundreds of seconds Telegram close the socket connection from the server side)

joroMaser, 2022-01-16:
@cotestatnt Thank you so much! last question please :
I send Contact and got the message"
```

{
  "ok": true,
  "result": [
    {
      "update_id": XXXXX,
      "message": {
        "message_id": 1067,
        "from": {
          "id": XXXXXXXX,
          "is_bot": false,
          "first_name": "XXXX",
          "language_code": "XX"
        },
        "chat": {
          "id": XXXXX,
          "first_name": "XXXXX",
          "type": "private"
        },
        "date": XXXXXXXX,
        "contact": {
          "phone_number": "XXXXXXXXXXXXX",
          "first_name": "JOJO",
          "vcard": "XXXXXXXXXXXXXXXXXXXXXXXXXXXX",
          "user_id": XXXXXXXXXXX
        }
      }
    }
  ]
}
```

And I want to print the details:

```
  if(msg.messageType==MessageContact)
  {
    Serial.println(msg.contact.firstName);  
  }
```
in output I see "J⸮?⸮J⸮?⸮J⸮? "
Why is that please?

cotestatnt, 2022-01-17:
Good morning @joroMaser 
I'm going to re-open this issue because this morning I've merged a [pull request ](https://github.com/cotestatnt/AsyncTelegram2/pull/36/commits/c41424b1437193efc63e86f2ae419dbda5dd6ada)from @neryaz regarding chat members in groups.

Now you can handle this kind of events, simply checking the type of message received

`msg.messageType == MessageNewMember `
 or
 `msg.messageType == MessageLeftMember`

If I have enought time today or in the nexts days, I will write a new example with this and some other features.

closed 2022-02-09 by joroMaser