cotestatnt/AsyncTelegram2 #33: error in library

unregistered33, 2022-01-16:
Thanks for library! But i have little problem (using esp32, core 1.0.6):
`AsyncTelegram2.cpp: 5:1: error: prototype for 'AsyncTelegram2::AsyncTelegram2(Client&)' does not match any in class 'AsyncTelegram2
 AsyncTelegram2*: AsyncTelegram2(Client &client)
 
AsyncTelegram2.cpp:1: In file included from
 
AsyncTelegram2.h: 85:7: error: candidates are: AsyncTelegram2::AsyncTelegram2(const AsyncTelegram2&)
   class AsyncTelegram2
 
AsyncTelegram2.h: 90:5: error:                 AsyncTelegram2::AsyncTelegram2(Client&, uint32_t)
   AsyncTelegram2(Client &client, uint32_t bufferSize = BUFFER_BIG)
 
AsyncTelegram2.cpp: 62:6: error: prototype for 'bool AsyncTelegram2::sendCommand(const char* const&, const char*, bool)' does not match any in class 'AsyncTelegram2
 bool AsyncTelegram2*: sendCommand(const char* const &command, const char* payload, bool blocking )
 
AsyncTelegram2.cpp:1: In file included from
 
AsyncTelegram2.h: 384:10: error: candidate is: bool AsyncTelegram2::sendCommand(const char*, const char*, bool)
   bool sendCommand(const char* command, const char* payload, bool blocking = false)
 
AsyncTelegram2.cpp: 344:6: error: prototype for 'bool AsyncTelegram2::forwardMessage(const TBMessage&, int32_t)' does not match any in class 'AsyncTelegram2
 bool AsyncTelegram2*: forwardMessage(const TBMessage &msg, const int32_t to_chatid)
 
AsyncTelegram2.cpp:1: In file included from
 
AsyncTelegram2.h: 157:10: error: candidate is: bool AsyncTelegram2::forwardMessage(const TBMessage&, int64_t)
   bool forwardMessage(const TBMessage &msg, const int64_t to_chatid)`

cotestatnt, 2022-01-16:
You are using a very old core, I strongly suggest to update it to the last 2.0.2

However the library compiles and works without problems even with the 1.0.6 core (just tested).

**Are you trying any of the examples?** 
If not, put the code that is causing you problems.

unregistered33, 2022-01-17:
My previous post is not a problem now. Errors occurred due to incorrect work of VisualMicro and libraries updates. And I'm switch to 2.0.2, but always have "Unable to connect to Telegram server". With 1.0.6, 2.0.0, 2.0.1 example work fine (lightBot)

cotestatnt, 2022-01-17:
Yes, you are rigth.
I've realized only this morning that with latest [ESP32 core for Arduino 2.0.2](https://github.com/espressif/arduino-esp32/releases/tag/2.0.2) WifiClientSecure doesn't work properly even with their included examples out of the box.

It's better stay on 2.0.1 release and wait for a bug fix.

Another option is enable the ClientSSL library for SSL encryption.

closed 2022-01-17 by unregistered33