cotestatnt/AsyncTelegram2 #34: Feedback about sending messages - successful or not successful.

SergGrn, 2022-01-16:
Hello.
Thank you very much for your work
AsyncTelegram2 and I'm very glad that there is no stopping the main code when working.
There is one important issue: when using myBot.sendTo(userid, welcome_msg) is there a feedback function that would return "true" to the controller if the message was successfully sent and "false" if the message was not sent?
For me it is very important that if the message was not sent, resend until it is successfully sent.
Thanks!

cotestatnt, 2022-01-16:
In order to check if message was succesfully delivered **you have to wait response** from Telegram server and this breaks the non-stopping working mode.
This is something I had already considered (indeed the `sendTo()` function is meant to returns a bool), but which I haven't worked on yet.
I have more or less an idea on how to proceed, but it is still to be tested.

For the moment I've done some quick fixes and you could solve this kind of workaround (after update library to latest release):
create a function calling sendMessage in blocking mode (send the messages and force waiting reply).
```
AsyncTelegram2 myBot(client);

bool sendToWait(const int64_t userid, const char* message, const char* keyboard = nullptr) {
  TBMessage msg;
  msg.chatId = userid;
  return myBot.sendMessage(msg, message, keyboard, true);
}
```

Then you can check if message was sent succesfully

```
......
    String msg = "Test message";
    if( sendToWait(userid, msg.c_str()) ) {
      Serial.println("Message sent");     
    }
    else {
      Serial.println("Message NOT sent");   
    }

    delay(2000);
    if( sendToWait(123456789, "Wrong id test message")) {
      Serial.println("Message sent");     
    }
    else {
      Serial.println("Message NOT sent");   
    }
```

SergGrn, 2022-01-16:
Do I understand correctly that such a solution will block the controller code for a while?
Since it is more important for me not to block the code (it interferes with other important processes), so if the code is blocked, then I will not check the sending of the message for now.

cotestatnt, 2022-01-16:
Yes a little.
I'm testing right now with a 10Hz blinking led in order to estimate timings and when ESP need to sent message i can measure  about 120/140ms of latency time (it depends from network speed also off course).
Red channel is tied to HIGH level just before sending message and LOW after evaluating the response.

```
    String msg = "Test message";
    digitalWrite(D2, HIGH);
    if( sendToWait(userid, msg.c_str()) ) {
      digitalWrite(D2, LOW);
      Serial.println("Message sent");     
    }
    else {
      Serial.println("Message NOT sent");   
    }
   
```

[image]

SergGrn, 2022-01-16:
I understand. Unfortunately, for me, the time of 140 ms can sometimes be critical. Therefore, I will not check the delivery of messages for the time being. Hope this can be fixed in the future.
Thank you very much for your help and input on my issue! Wish you success.

cotestatnt, 2022-01-17:
Hi @SergGrn.
I'm testing a new functionality of library born from your issue.
In order to handle the response from server after sent a new message, I've added the capabilities to run a callback function (off course only if defined) to get back the result of message delivery completely asynchronus.

[sendMessageCallback.ino](https://github.com/cotestatnt/AsyncTelegram2/blob/main/examples/sendMessageCallback/sendMessageCallback.ino)

Aniway, regarding your specific needs, if timings are really really critical, i would suggest you to use ESP32 and run Telegram stuffs as a distinct parallel task thanks to the support for FreeRTOS. I can provide you a working example code, if you like.

SergGrn, 2022-01-17:
Hello.
Once again, thank you very much for your help and participation in my questions.
I will definitely try using sendMessageCallback.ino.
Yes, I use exactly ESP32. So far, the work of the library suits me quite well, so I did not think about FreeRTOS. Although I might need it. I will think.
Tomorrow I will try to connect sendMessageCallback.ino.
Thanks again!

SergGrn, 2022-01-18:
> Hi @cotestatnt 
For some reason, the sendMessageCallback.ino example does not compile for me.
Compile error: 'class AsyncTelegram2' has no member named 'addSentCallback'

cotestatnt, 2022-01-18:
Sorry @SergGrn I have released the latest version just now 2.0.7

SergGrn, 2022-01-21:
Hi @cotestatnt.
I found something odd in the sendMessageCallback.ino function.
I turn off the Internet (wifi), after that I make a one-time attempt to send a message to the server using the sendTo() function. Then, after a few minutes, I connect the Internet, while the sendMessageCallback.ino function returns "true" - that the message was delivered to the server, but in fact the message, of course, did not reach the server and did not appear in the chat, since it was sent at the time when there was no internet connection. It turns out a false message "true" about the successful sending of the message...

cotestatnt, 2022-01-22:
Thanks @SergGrn for your detailed feedback.
I think a timeout is needed, or even better, timeout and feedback about specific message id.
I will wok on soon

cotestatnt, 2022-01-23:
@SergGrn in the latest release, after send a message from bot, the library will look forward for feedback about the specific **message id** just sent. 
In addition, a configurable timeout was implemented with a defualt time of 5 seconds.

  ```
  // Add the callback function to bot
  // myBot.addSentCallback(messageSent);  // Default 5000 milleseconds
  
  myBot.addSentCallback(messageSent, 2000);  // custom time of 2000 milliseconds
```

SergGrn, 2022-01-24:
@cotestatnt, thanks for the help!

closed 2022-02-02 by cotestatnt