cotestatnt/AsyncTelegram2 #38: sendMessage reply to chatId from incoming message by default

dsultanr, 2022-01-18:
when I sending a message to a bot on the group the bot answers directly to the sender in private message but not to the group
seems bot answers to the senderId instead of chatId
is it an issue?

cotestatnt, 2022-01-19:
It's not properly an issue. Up now is not handled if the message was received from a group or from an user. I'll take in consideration for the next release.
Anyway you can easily achieve your needs in this way:

```
int64_t groupid = -123456789;
..................

if (myBot.getNewMessage(msg)) {

   msg.chatId = groupid;
   msg.sender.id = groupid;
   ........
```

dsultanr, 2022-01-19:
I did replace the sendMessage call to the sendTo call as a workaround in your keyboards.ino example
so the bot now answers to the chat that is received the message - either direct or group

        // check if is show keyboard command
        if (msgText.equalsIgnoreCase("/reply_keyboard")) {
          // the user is asking to show the reply keyboard --> show it
          // myBot.sendMessage(msg, "This is reply keyboard:", myReplyKbd); // old behavior
          myBot.sendTo(msg.chatId, "This is reply keyboard:", myReplyKbd.getJSON());
          isKeyboardActive = true;
        }

closed 2022-01-21 by cotestatnt