cotestatnt/AsyncTelegram2 #41: Yes i found that library missing,

muk3shb, 2022-01-21:
How can we implement TOTP on our bot ? For that there is TOTP library in Arduino and then it is saying to generate "hmac" key. or how can we limit the IP access like telegram_ip_lower or telegram_ip_upper are there any such function under Async library to limit the IP's ? 

Next, can we limit the ESP32 board to sepcific IP's ? as telegram bots can be accessed by any user around the world hence, all we can do is to limit the IP access or is there any other method ?

droidprova, 2022-01-21:
Hi, if you look at the examples, in the setup you find this:

  ```
 int64_t user id = 123456789;
 myBot.sendTo (userid, welcome_msg);
```

this means that only the telegam user with id = 123456789 will be able to receive replies from your bot. Similarly, you can do the same by telling your bot to receive and process responses only from one or more selected telegram ID . 

Hello

muk3shb, 2022-01-21:
> Hi, if you look at the examples, in the setup you find this:
>
> ```
> int64_t user id = 123456789;
> myBot.sendTo (userid, welcome_msg);
> ```
>
> this means that only the telegam user with id = 123456789 will [...]

Thankyou So much!! I found that code on the example and with little changes, it is working!! i have not changed anything just very minute changes. There are no errors but the message is continuously generating. Here is the code.. 

// i am reverifying the chatid by using simple operator and it is working!! but the bot is continuosly generating the message. the code that i have mentioned will work fine

#define USE_CLIENTSSL false  

#include <AsyncTelegram2.h>
// Timezone definition
#include <time.h>
#define MYTZ "CET-1CEST,M3.5.0,M10.5.0/3"

#ifdef ESP8266
  #include <ESP8266WiFi.h>
  BearSSL::WiFiClientSecure client;
  BearSSL::Session   session;
  BearSSL::X509List  certificate(telegram_cert);
  
#elif defined(ESP32)
  #include <WiFi.h>
  #include <WiFiClient.h>
  #if USE_CLIENTSSL
    #include <SSLClient.h>  
    #include "tg_certificate.h"
    WiFiClient base_client;
    SSLClient client(base_client, TAs, (size_t)TAs_NUM, A0, 1, SSLClient::SSL_ERROR);
  #else
    #include <WiFiClientSecure.h>
    WiFiClientSecure client;    
  #endif
#endif

AsyncTelegram2 myBot(client);
const char* ssid  =  "xxxxxxxx";     // SSID WiFi network
const char* pass  =  "xxxxxxxxxx";     // Password  WiFi network
const char* token =  "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx";  // Telegram token

const int LED = 5;

void setup() {
  // initialize the Serial
  Serial.begin(115200);
  Serial.println("\nStarting TelegramBot...");

  // set the pin connected to the LED to act as output pin
  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW); // turn off the led (inverted logic!)

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  delay(500);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(500);
  }
  

#ifdef ESP8266
  // Sync time with NTP, to check properly Telegram certificate
  configTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
  //Set certficate, session and some other base client properies
  client.setSession(&session);
  client.setTrustAnchors(&certificate);
  client.setBufferSizes(1024, 1024);
#elif defined(ESP32)
  // Sync time with NTP
  configTzTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
  #if USE_CLIENTSSL == false
    client.setCACert(telegram_cert);
  #endif
#endif

  // Set the Telegram bot properies
  myBot.setUpdateTime(1000);
  myBot.setTelegramToken(token);

  // Check if all things are ok
  Serial.print("\nTest Telegram connection... ");
  myBot.begin() ? Serial.println("OK") : Serial.println("NOK");

  Serial.print("Bot name: @");
  Serial.println(myBot.getBotName());

  char welcome_msg[128];
  snprintf(welcome_msg, 128, "BOT @%s online\n/help all commands avalaible.", myBot.getBotName());
  int64_t chat_id = 1234567890; // You can discover your own chat id, with "Json Dump Bot"
  myBot.sendTo(chat_id, welcome_msg);

   
}

void loop() {
  // local variable to store telegram message data
  TBMessage msg;
  // if there is an incoming message...
  if( myBot.getNewMessage(msg)); {
    String msgText = msg.text;

    if (msgText.equals("/light_on")) {                 // if the received message is "LIGHT ON"...
      digitalWrite(LED, HIGH);                          // turn on the LED (inverted logic!)
      myBot.sendMessage(msg, "Light is now ON");       // notify the sender
    }
    else if (msgText.equals("/light_off")) {           // if the received message is "LIGHT OFF"...
      digitalWrite(LED, LOW);                          // turn off the led (inverted logic!)
      myBot.sendMessage(msg, "Light is now OFF");       // notify the sender
    }
    else {                                              // otherwise...
      // generate the message for the sender
      String reply;
      reply = "Welcome " ;
      reply += msg.sender.username;
      reply += ".\nTry /light_on or /light_off ";
      myBot.sendMessage(msg, reply);                    // and send it
    }
  }

}

cotestatnt, 2022-01-23:
@muk3shb please use code tag <> for sketches.

I've replied in the other [issue ](https://github.com/cotestatnt/AsyncTelegram2/issues/42)opened by you, but it's not clear is this issue is the same or not.

muk3shb, 2022-01-23:
> @muk3shb please use code tag <> for sketches.
> 
> I've replied in the other [issue ](https://github.com/cotestatnt/AsyncTelegram2/issues/42)opened by you, but it's not clear is this issue is the same or not. 

Thankyou sir, This issue is the same and I was working on this and i thought that the Library should have some updated version and You now here is the Latest Update by you. Thankyou So Much sir.

cotestatnt, 2022-01-23:
Ok, i'm going to close both issue then.

Thanks for your feedback

closed 2022-01-23 by cotestatnt

muk3shb, 2022-01-23:
> Ok, i'm going to close both issue then.
> 
> Thanks for your feedback

Your Welcome Sir.