cotestatnt/AsyncTelegram2 #43: esp8266 is blocked

phazz1980, 2022-01-23:
when receiving invalid JSON, ESP is blocked for about 15 minutes

```
00:13:45.169 -> [E][AsyncTelegram2.cpp:183] getNewMessage():	deserializeJson() failed
00:13:45.169 -> 
00:13:45.169 -> 
00:13:45.169 -> [E][AsyncTelegram2.cpp:184] getNewMessage():	IncompleteInput
00:13:45.169 -> {"ok":true,"result":{"message_id":16672,"from":{"id":2056******,"is_bot":true,"first_name":"smpBoil","username":"smpBoilBot"},"chat":{"id":-611******,"title":"Test \u0410\u041d\u0414-02","type":"group","all_members_are_administrators":true},"date":1642965225,"text":"\u2139 \u0410\u0432\u0430\u0440\u0438\u044f \u043d\u0438\u0437\u043a\u043e\u0439 \u0442\u0435\u043c\u043f\u0435\u0440\u0430\u0442\u0443\u0440\u044b! \u2744\n\ud83c\udf21\u0422\u0435\u043c\u043f\u0435\u0440\u0430\u0442\u0443\u0440\u0430 \u043a\u043e\u0442\u043b\u0430: 0.0\u00b0\u0421\n\ud83d\udd25\u041c\u0430\u043a\u0441.\u0442\u0435\u043c\u043f\u0435\u0440\u0430
00:28:45.226 -> 
00:28:45.226 -> 
00:28:45.226 -> [E][AsyncTelegram2.cpp:156] getUpdates():	
00:28:47.240 -> [D][AsyncTelegram2.cpp:28] checkConnection():	Start handshaking...
00:28:51.632 -> [D][AsyncTelegram2.cpp:34] checkConnection():	Connected using Telegram hostname
00:28:51.632 -> Last connection was 995 seconds ago
```

cotestatnt, 2022-01-24:
Hi @phazz1980 
I'm trying to reproduce your issue, but i don't undestand how you can get a such long list of unicode entities instead normal text.
`\u00b0\u0421\n\ud83d\udd25\u041c\u0430\u043a\u0441.\u0442\u0435\u043c\u043f\u0435\u0440 etc etc`

In this way the message is much longer than necessary because you need 6 bytes for each charater and buffer allocated memory is not enought for JSON parsing.

I've tryed also to switch to Russian in my app but the message received in serial monitor (enabling debug functionality) is still without unicode entities.

```
     .................
      "type": "private"
        },
        "date": 1643016127,
        "text": "ℹ️ Авария низкой температуры! ❄️\n🌡Температура котла: 0.0°С\n🔥Макс.темпера"
      }
```

Off course, the bug is that error is not properly handled and ESP freeze. I'm working on it, but in the meantime if you can understand why you got unicode entities instead normal text, I think you will avoid this malfunction.

phazz1980, 2022-01-24:
> Off course, the bug is that error is not properly handled and ESP freeze.

Yes. This problem also occurs if the telegram api blocks the bot due to exceeding the number of messages

cotestatnt, 2022-01-24:
Hi @phazz1980 
I've just  [pushed a fix](https://github.com/cotestatnt/AsyncTelegram2/commit/a08c73486653682936eabc9a010ebe8bd5421328) for error handling (still not released, it will be included in next release, but you can edit yourself source code or directly download zip file from Github).

Now MCU won't freeze and a message is sent informing user that message is too long to be parsed.

If you still need to handle such type of very long messages, you should increase the size of `#define BUFFER_BIG  2048` inside [DataStructures.h](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/DataStructures.h#L5). 
I can't increase it by default, because is unuless more memory consumption which normally is not needed

closed 2022-02-02 by cotestatnt