cotestatnt/AsyncTelegram2 #46: ESP8266 reboots

gravedigger21, 2022-02-01:
I use your library AsyncTelegram2.h
Uploaded the lightBot.ino example. Through telegram, I turned on the LED and left it on. After a while I noticed that it went out. I turned it on again, but the situation repeated itself. Then I started the Serial Monitor and saw that after about 30-50 minutes the ESP8266 reboots with an Exception (2) error:
10:36:19.176 -> epc1=0x3ffed570 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffed570 depc=0x00000000
or Soft WDT reset.
Here is the decoding of EspExceptionDecoder:

`Exception 2: InstructionFetchError: Processor internal physical address or data error during instruction fetch
PC: 0x3ffed570
EXCVADDR: 0x3ffed570

Decoding stack results
0x4020a61c: String::equals(char const*) const at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 505
0x402226bc: sha2big_out at src/inner.h line 582
0x40222414: sha2big_round at src/hash/sha2big.c line 131
0x40222c9b: br_sha2small_round at src/hash/sha2small.c line 113
0x4023125d: br_ghash_ctmul32 at src/inner.h line 585
0x40224669: br_ssl_engine_set_buffers_bidi at src/ssl/ssl_engine.c line 442
0x40225a59: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 151
0x40222b8f: br_sha2small_round at src/hash/sha2small.c line 105
0x40231ca1: br_i15_reduce at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h line 77
0x40223598: br_i15_montymul at src/int/i15_montmul.c line 39`

Decoding Soft WDT reset:
`Decoding stack results
0x40208b16: BearSSL::WiFiClientSecure::_clearAuthenticationSettings() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 98
0x402249aa: br_ssl_engine_recvrec_ack at src/ssl/ssl_engine.c line 743
0x40208d30: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 436
0x40208f13: BearSSL::WiFiClientSecure::_write(unsigned char const*, unsigned int, bool) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 287
0x4020a6a8: String::indexOf(String const&, unsigned int) const at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 659
0x4020acf6: __wrap_system_restart_local() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_postmortem.cpp line 212
0x40207b08: ClientContext::_write_from_source(DataSource*) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/include/ClientContext.h line 443
0x40100294: __digitalWrite(uint8_t, uint8_t) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring_digital.cpp line 85
0x40209031: BearSSL::WiFiClientSecure::_wait_for_handshake() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 562
0x40201a97: AsyncTelegram2::getUpdates() at C:\Users\Admin\Documents\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 116
0x40201b35: AsyncTelegram2::getUpdates() at C:\Users\Admin\Documents\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 136
0x40207bc0: WiFiClient::write_P(char const*, unsigned int) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/include/DataSource.h line 57
0x40100294: __digitalWrite(uint8_t, uint8_t) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring_digital.cpp line 85
0x40202b08: ArduinoJson6191_F1::JsonDeserializer  , ArduinoJson6191_F1::StringCopier>::parseVariant (ArduinoJson6191_F1::VariantData&, ArduinoJson6191_F1::AllowAllFilter, ArduinoJson6191_F1::NestingLimit) at C:\Users\Admin\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp line 66
0x40100228: millis() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40202de3: ArduinoJson6191_F1::JsonDeserializer  , ArduinoJson6191_F1::StringCopier>::parseObject (ArduinoJson6191_F1::CollectionData&, ArduinoJson6191_F1::AllowAllFilter, ArduinoJson6191_F1::NestingLimit) at C:\Users\Admin\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp line 416
0x40202b08: ArduinoJson6191_F1::JsonDeserializer  , ArduinoJson6191_F1::StringCopier>::parseVariant (ArduinoJson6191_F1::VariantData&, ArduinoJson6191_F1::AllowAllFilter, ArduinoJson6191_F1::NestingLimit) at C:\Users\Admin\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp line 66
0x40100294: __digitalWrite(uint8_t, uint8_t) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring_digital.cpp line 85
0x40100a3f: realloc(void*, size_t) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 619
0x40204bf2: AsyncTelegram2::getNewMessage(TBMessage&) at C:\Users\Admin\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp line 107
0x40201332: _GLOBAL__sub_I_client() at C:\Users\Admin\Documents\Arduino\libraries\AsyncTelegram2\examples\lightBot/lightBot.ino line 125
0x40201332: _GLOBAL__sub_I_client() at C:\Users\Admin\Documents\Arduino\libraries\AsyncTelegram2\examples\lightBot/lightBot.ino line 125
0x40209480: BearSSL::CertStore::findHashedTA(void*, void*, unsigned int) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp line 197
0x4020948c: BearSSL::CertStore::findHashedTA(void*, void*, unsigned int) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp line 200
0x40209480: BearSSL::CertStore::findHashedTA(void*, void*, unsigned int) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp line 197
0x40209751: HardwareSerial::peek() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 136
0x402099cd: IPAddress::printTo(Print&) const at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\IPAddress.cpp line 154
0x4020b6b9: uart_read(uart_t*, char*, size_t) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 266
0x40100294: __digitalWrite(uint8_t, uint8_t) at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring_digital.cpp line 85
0x40209e2d: Stream::timedRead() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Stream.cpp line 38
0x40100196: esp_schedule() at C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 125
0x402011ff: loop() at C:\Users\Admin\Documents\Arduino\libraries\AsyncTelegram2\examples\lightBot/lightBot.ino line 109`

cotestatnt, 2022-02-01:
Hi @gravedigger21 
I will try to reproduce the same behaviour, but as first point I would suggest you to update to the latest [ESP8266 Core for Arduino](https://github.com/esp8266/Arduino )because you are using a very old major realase 2.7.4 while the latest stable is 3.0.2 with tons of improvements.

gravedigger21, 2022-02-01:
Hi @cotestatnt 
Many of my old projects don't work with version 3.x.x. So I'm staying on 2.7.4 for now. But I tried to update on another PC to version 3.0.2 and the problem seems to be gone. ESP8266 worked without reboot for more than 2 hours. But when I updated the core, I also updated the ArduinoJson library to the latest version (6.191). and another problem appeared with it - frequent messages "Unable to connect to Telegram server" began to appear in the port monitor. After reverting to older versions - 6.17.3 or 6.18.0, these messages are gone.

cotestatnt, 2022-02-02:
Thanks @gravedigger21 for your feedback.
You shoul try to use as IDE [VSCode ](https://code.visualstudio.com/)+ [Platformio](https://platformio.org/) because you can set versions for both MCU core and libraries for each project you have (inside the file platformio.ini) so you don't have to lose time with old or new versions every time you need to compile again.

phazz1980, 2022-02-05:
`
Decoding stack results
0x4021030e: optimistic_yield(uint32_t) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/core_esp8266_features.h line 66
0x4020ab10: BearSSL::WiFiClientSecureCtx::_run_until(unsigned int, bool) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/PolledTimeout.h line 234
0x4022d534: br_ssl_hs_client_init_main at src/ssl/ssl_hs_client.c line 907
0x4022d584: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 960
0x4022cb72: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1319
0x4020ac2f: BearSSL::WiFiClientSecureCtx::_wait_for_handshake() at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 585
0x4020ae1f: BearSSL::WiFiClientSecureCtx::_connectSSL(char const*) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 1170
0x4021023c: __esp_yield() at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 116
0x4021093e: __delay(unsigned long) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40209d3a: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/include/ClientContext.h line 148
0x4020aec1: BearSSL::WiFiClientSecureCtx::connect(char const*, unsigned short) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 231
0x40212246: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at c:\users\�����\desktop\flprog7.4\idev8\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits/shared_ptr_base.h line 1324
0x40202d84: AsyncTelegram2::checkConnection() at c:\Users\�����\Desktop\Flprog7.4\ideV8\portable\sketchbook\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 41
0x40202e13: AsyncTelegram2::sendCommand(char const*, char const*, bool) at c:\Users\�����\Desktop\Flprog7.4\ideV8\portable\sketchbook\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 64
0x40100325: millis() at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 193
0x4020304c: AsyncTelegram2::getUpdates() at c:\Users\�����\Desktop\Flprog7.4\ideV8\portable\sketchbook\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 121
0x40205d9a: AsyncTelegram2::getNewMessage(TBMessage&) at c:\Users\�����\Desktop\Flprog7.4\ideV8\portable\sketchbook\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 178
0x40205d6c: AsyncTelegram2::getNewMessage(TBMessage&) at c:\Users\�����\Desktop\Flprog7.4\ideV8\portable\sketchbook\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 172
0x40204440: ArduinoJson6183_D1::deserializeJson (ArduinoJson6183_D1::JsonDocument&, String&, ArduinoJson6183_D1::NestingLimit) at c:\Users\�����\Desktop\Flprog7.4\ideV8\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp line 32
0x40100325: millis() at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 193
0x40205d9d: AsyncTelegram2::getNewMessage(TBMessage&) at c:\Users\�����\Desktop\Flprog7.4\ideV8\portable\sketchbook\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 178
0x40100210: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100210: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100210: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100210: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100210: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40217f52: memmove_P at /workdir/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c line 189
0x401008db: umm_free_core(umm_heap_context_t*, void*) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 549
0x40100d14: realloc(void*, size_t) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1166
0x40100ae4: malloc(size_t) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 821
0x40100b1c: realloc(void*, size_t) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 853
[... cut ...]
0x40100210: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x4021035c: loop_wrapper() at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 201
0x401000c8: app_entry() at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 331
0x40232200: poly1305_inner at src/symcipher/poly1305_ctmul32.c line 60
0x40220b22: tcp_split_unsent_seg at core/tcp_out.c line 966
`

cotestatnt, 2022-02-06:
Thanks @phazz1980 
It seems that sometimes SSL layer fail to connect to server and then ESP watchdog get triggered.
I will try to handle better this kind of situation in order to avoid this.
It's not quick process unluckily because in my setup it happens more or less once a day. I will try with a wifi network with poor signal.

phazz1980, 2022-02-08:
this library cannot be used without SSL in any way?

cotestatnt, 2022-02-08:
Telegram server accept connection only on port 443 with encrypted data (TLS/SSL).

Any way, certificate validation trust chain can be avoided on ESP side simply enabling the "setInsecure()" option for WiFiClientSecure.
The connection is still encrypted, [but certicate verification will be bypassed](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html#setinsecure). 
Even if it is not the best, it could be an acceptable work-around for now because in any case the address to connect to (api.telegram.org) is encoded in the firmware and no one has a way to change it once compiled. 

  ```
//client.setSession(&session);
  //client.setTrustAnchors(&certificate);
  //client.setBufferSizes(TCP_MSS, TCP_MSS);
  client.setInsecure();
```

joroMaser, 2022-02-09:
@cotestatnt Are you sure that watchdog reset is because SSL layer fail to connect to server?
If I remember right the library will connect each 300 seconds to Telegram server , so 1 time in day that connect is failed and watchdog is reset?

phazz1980, 2022-02-10:
> Telegram server accept connection only on port 443 with encrypted data (TLS/SSL).
>
> Any way, certificate validation trust chain can be avoided on ESP side simply enabling the "setInsecure()" option [...]

no connection (

cotestatnt, 2022-02-28:
This issue is very puzzling for me. 
It's seem related to library `checkConnection()` method causing sometime esp8266 soft WDT reset.
Moving directly the same instructions used in that method where needed, the reset problem seems disappear but I can't figure why..

Anyway, I will test again for some time and if everything's ok I'll commit the new sources and draft a new release.

cotestatnt, 2022-03-01:
No success, I'm still facing with WDT reset

joroMaser, 2022-03-01:
@cotestatnt 
Are you able to recover the reboot consistently?
Can you copy here you code?

joroMaser, 2022-03-01:
@cotestatnt I running you last commit and after 1 hours I got :

```

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3ffff5c0 end: 3fffffb0 offset: 01a0
3ffff760:  6fe7640e 4bc6a7f0 00000000 00003a98  
3ffff770:  6fe7640e 00000000 00418937 00000000  
3ffff780:  000000fd 000000fd 401002a6 00000002  
3ffff790:  6fe7640e 3ffff7c0 3fff2cac 4020a23a  
3ffff7a0:  00000000 3fff3f54 3ffefcbc 00003a98  
3ffff7b0:  00000000 0003a475 3ffefcbc 4020860c  
3ffff7c0:  000000fd 79c3b77c 742470de d49d3dfb  
3ffff7d0:  00000008 00000001 009fb0af 00000000  
3ffff7e0:  40226b2c 40226b7c 3fff2cac 4022616a  
3ffff7f0:  3fff31ac 3ffe8adb 3fff2cac 3ffe8adb  
3ffff800:  00000000 00000001 3ffefcbc 4020872b  
3ffff810:  00000000 00000001 3ffefcbc 4020891b  
3ffff820:  00000052 3ffef15c 00000001 4020a168  
3ffff830:  00000000 00000000 000001bb 4020a86a  
3ffff840:  00000000 3fff1abc 3ffefcbc 4020786e  
3ffff850:  3fff18d4 000001ff 3ffeef3c 3ffe8c92  
3ffff860:  000001bb 3ffe8adb 3ffefcbc 3ffe8c92  
3ffff870:  000001bb 3ffe8adb 3ffefcbc 402089bd  
3ffff880:  4020d9fc dca79a95 4020d9fc dca79a95  
3ffff890:  3ffff8f0 00000000 3ffeee40 4020c9e2  
3ffff8a0:  00000002 00000000 3ffff8d0 40201d38  
3ffff8b0:  0000000c 00000000 401002c1 00000002  
3ffff8c0:  00000000 3ffefcbc 3ffeee40 29c3c2c9  
3ffff8d0:  978d4fe0 00000000 3ffeee40 00000000  
3ffff8e0:  3fffdad0 00000000 3ffeee40 40201f94  
3ffff8f0:  696c227b 2274696d 222c313a 656d6974  
3ffff900:  2274756f 222c303a 7366666f 3a227465  
3ffff910:  36303037 34323539 3f007d31 40102b68  
3ffff920:  00000005 00000000 00000020 401001ac  
3ffff930:  00000001 6eca29d0 00000005 40101e00  
3ffff940:  3ffe97b5 40105097 3ffed218 401001ac  
3ffff950:  4010298b 3ffed218 00000002 40101e00  
3ffff960:  00000001 6fb489bd 3ffedad4 40102b68  
3ffff970:  3ffea06c 00000000 00000000 40105911  
3ffff980:  00000000 00000000 0000001f 401001ac  
3ffff990:  3ffea06c 7fffffff 3fffc228 40105911  
3ffff9a0:  40103f8f 00040000 00000000 00040000  
3ffff9b0:  53000000 40103f8c 00040000 401001ac  
3ffff9c0:  3ffed930 40102ef3 3fffc228 40105911  
3ffff9d0:  4000050c 6c6f57a0 00000000 fffffffe  
3ffff9e0:  401002b0 2c9f0300 4000050c 3fffc278  
3ffff9f0:  40102d44 3fffc200 00000022 00000000  
3ffffa00:  00000000 00000009 3ffffa70 402030d5  
3ffffa10:  00000001 00000008 3fff43b7 40201f05  
3ffffa20:  3fff43b7 978d4fe0 00000000 0000000a  
3ffffa30:  3ffffc50 00000000 3ffffa70 40203246  
3ffffa40:  00000001 00000000 3fff43b7 00000030  
3ffffa50:  4010305c 00080000 00000010 00000000  
3ffffa60:  00000000 00000000 3ffffc38 40203390  
3ffffa70:  3ffffc38 3fff43b7 00000007 000007ed  
3ffffa80:  40102d01 3fff20cb 3fff20cb 0000007d  
3ffffa90:  3ffffc38 65757274 00000000 ffffffff  
3ffffaa0:  00000002 4bc6a7f0 7374bc6a 2116577d  
3ffffab0:  00000002 00000000 4bc6a7f0 00000000  
3ffffac0:  ffffffff 00000000 401002c1 00000002  
3ffffad0:  001c9db3 00000020 3fff43b4 40100996  
3ffffae0:  3ffe8ada 3ffffe40 3ffffc38 3ffef0f4  
3ffffaf0:  3fffdad0 3ffffe40 00000000 40204d0d  
3ffffb00:  3fff4bcc 00000000 3ffffc38 3ffe8cca  
3ffffb10:  00000000 3ffe8def ca7ef9db 151fa3ea  
3ffffb20:  00000000 00000000 4bc6a7f0 00000000  
3ffffb30:  3fff1abc 000001b9 401002c1 00000000  
3ffffb40:  00000000 00000000 3ffefcbc 4020a206  
3ffffb50:  00000000 3ffefcbc 3ffefcbc 402260fe  
3ffffb60:  00000000 00000001 3ffefcbc 402086e9  
3ffffb70:  00000001 3ffffbb0 3ffefcbc 40207e64  
3ffffb80:  00000000 00000000 0000001f 401001ac  
3ffffb90:  00000965 00000965 3fffc228 40105911  
3ffffba0:  4000050c 00000001 3ffeee40 3ffe8c0c  
3ffffbb0:  401015b7 00000030 00000019 fffffffe  
3ffffbc0:  0000000f 00000003 0000006b 00000080  
3ffffbd0:  00000000 00000001 00000080 fffffffe  
3ffffbe0:  00000340 3ffea048 00000000 60009600  
3ffffbf0:  00000000 00000000 0000001f 401001ac  
3ffffc00:  dde3d260 0a1bb067 3fffc228 40105911  
3ffffc10:  4000050c 3ffeee40 3ffeee40 40201e98  
3ffffc20:  40000502 00000030 0000001f fffffffe  
3ffffc30:  000005e0 00000000 3fff43b4 3fff43be  
3ffffc40:  3fff4b94 3fff4bb4 40103e00 00000100  
3ffffc50:  3fff4ba4 3fff4b94 40103e20 3ffed930  
3ffffc60:  00000005 00000000 00000020 401001ac  
3ffffc70:  00000005 00000000 00000005 40101e00  
3ffffc80:  00000005 00000000 00000020 401001ac  
3ffffc90:  4010298b 3ffed218 00000005 40101e00  
3ffffca0:  3ffe97b5 40105097 3ffed218 40102b68  
3ffffcb0:  4010298b 3ffed218 00000000 40102b68  
3ffffcc0:  00000003 6fc109c5 3ffedad4 40102b68  
3ffffcd0:  3ffea048 00000000 00000000 00000001  
3ffffce0:  00000003 6fc109c5 40103022 00000100  
3ffffcf0:  3ffea048 7fffffff 00000000 00000001  
3ffffd00:  00000000 00000000 0000001f 401001ac  
3ffffd10:  00000000 00000000 0000001f 401001ac  
3ffffd20:  4000050c 2c9f0300 3fffc228 40105911  
3ffffd30:  4000050c 00000030 00000010 ffffffff  
3ffffd40:  4020cb32 00000030 00000010 ffffffff  
3ffffd50:  40201428 3ffffe40 3ff00014 00000000  
3ffffd60:  00000000 00000000 00⸮ծ⸮⸮⸮⸮3⸮
```

joroMaser, 2022-03-02:
@cotestatnt I running your last commit for few hours and I have got few reboot after print log `Start connection handshaking..`: 

Software reset 08:53 (`ctx: sys sp: 3ffff5c0 end: 3fffffb0 offset: 01a0` ), 10:33  ( `ctx: sys sp: 3ffff610 end: 3fffffb0 offset: 01a0` )
Hardware reset : 14:29
Software reset 16:15 (` ctx: cont sp: 3ffff5f0 end: 4010298b offset: 01a0 ` ) ,  18:10 ( `ctx: sys sp: 3ffff5f0 end: 3fffffb0 offset: 01a0` ) 

Do you have any idea how to print some log that will help you to understand that ?

cotestatnt, 2022-03-08:
Hello to everyone.
I have an ESP-01 running without reboots since a couple of days.
Could someone give me some feedback about latest release?

Thanks in advance.

joroMaser, 2022-03-10:
@cotestatnt I run your last commit and saw this log

```

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 111 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 23 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 148 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 27 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 21 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 29 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 31 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 27 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 52 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 26 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 22 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 53 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 34 seconds ago

[D][AsyncTelegram2.cpp:63] reset():	Restart Telegram connection

[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x4e9a335d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: bearssl
sp: 3fff1528 end: 3fff17b8 offset: 0190

...........
ctx: cont
sp: 3ffff920 end: 3fffffc0 offset: 0000
........
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

joroMaser, 2022-03-10:
I have got an exception again

```

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 294 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 302 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 301 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 299 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 300 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 312 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 300 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 302 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 300 seconds ago

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:43] checkConnection():	Connected using Telegram hostname
Last connection was 300 seconds ago

[D][AsyncTelegram2.cpp:234] getNewMessage():	
{
  "ok": true,
  "result": [
    {
      "update_id": XXXXXXXXX,
      "message": {
        "message_id": XXX,
        "from": {
          "id": XXXXXXXXX,
          "is_bot": false,
          "first_name": "XXXXXXXXX",
          "language_code": "en"
        },
        "chat": {
          "id": -XXXXXXXXX,
          "title": "XXXXXXXXX",
          "type": "supergroup"
        },
        "date": XXXXXXXXX,
        "text": "Ggg"
      }
    }
  ]
}

[D][AsyncTelegram2.cpp:419] sendMessage():	
{
  "chat_id": -XXXXXXXXX,
  "text": "Hi",
  "parse_mode": "HTML",
  "reply_markup": {
    "inline_keyboard": [
      [
        {
          "text": " press here!",
          "callback_data": "press1"
        }
      ]
    ]
  }
}

[D][AsyncTelegram2.cpp:163] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:29] checkConnection():	Start handshaking...

Unable to connect to Telegram server

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x3e77b67c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: bearssl
sp: 3fff12e8 end: 3fff17b8 offset: 0190
```

joroMaser, 2022-03-12:
@cotestatnt I will be happy to get your comment about the 2 exceptions from  bearssl. 

Anyway I run your last commit about 72 hours without reboot. What did you change that prevent watchdog reboot?

silvije2, 2022-03-13:
with asyncTelegram2 ver 2.0.8 my code was fairly stable, esp reseted once a day, after update to ver 2.1.0 code keeps reseting every minute, probably due to memory

so I moved some large static char arrays to global scope and esp stopped reseting but now client have problem getting new message

after enabling debug this is what I got:

[E][AsyncTelegram2.cpp:202] getNewMessage():	deserializeJson() failed
[E][AsyncTelegram2.cpp:203] getNewMessage():	NoMemory

also moving from char arrays to memory consuming Strings suprised me a lot

phazz1980, 2022-03-14:
```
Decoding stack results
0x40212a4c: BearSSL::WiFiClientSecureCtx::_run_until(unsigned int, bool) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 474
0x4023a0a0: br_ssl_hs_client_init_main at src/ssl/ssl_hs_client.c line 907
0x40212a4c: BearSSL::WiFiClientSecureCtx::_run_until(unsigned int, bool) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 474
0x4023a0a0: br_ssl_hs_client_init_main at src/ssl/ssl_hs_client.c line 907
0x4023a0f0: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 960
0x402396de: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1319
0x4023a0f0: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 960
0x402396de: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1319
0x40212b57: BearSSL::WiFiClientSecureCtx::_wait_for_handshake() at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 585
0x40212d47: BearSSL::WiFiClientSecureCtx::_connectSSL(char const*) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp549
0x40100cd4: realloc(void*, size_t) at C:\Users\�����\Desktop\Flprog7.4\ideV8\portable\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1166
```
reboot occurs at the moment of connecting to Wi-Fi or immediately connecting the router to the Internet

konig87nikkinakki, 2022-06-23:
i had the same problem with ESP8266, and i also tried other libraries: sooner or later it happened again.
So i moved to ESP32 (days to wait after buy to deliver, bad sleeping, i hate to wait for days) but with ESP32 library works reeeeally fine, i didnt have problems yet.

closed 2023-03-15 by cotestatnt