cotestatnt/AsyncTelegram2 #47: Few questions about AsyncTelegram2::getNewMessage

joroMaser, 2022-02-08:
1- Do you get each time 1 message only and the next message will come in the next loop?

2- https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.cpp#L190 
Please make 12 as constant strlen(""update_id":")

3- Why do you comment out 
https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.cpp#L198

4- Why do you use "6" 
https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.cpp#L192
Can you please do that with constant value with name ?

cotestatnt, 2022-02-08:
- 1: Yes. One message at once so it's easyier handle it.
- 2:  I did not understand the request. 12 is the length of the string searched for in the stream. Even if you put the `strlen()` statement, it would be optimized by the compiler. 
- 3: It's not necessary close the connection. Message was recieved, but can't be parsed due to the limitation of JSON object (in order to keep low SRAM consumption, if neeeded user can increase).
- 4: Same as point 2 but for a shorter pattern string
- 5: Soon, i'm trying to fix an annoying software watchdog reset with ESP8266 (even if it's seem not related 100% to library)

Inline_keyboard: 
yes, when you click on a inline button, Telegram will wait for a "endQuery()" response.
Check the example [keyboards.ino](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/keyboards/keyboards.ino#L199): once you get a MessagQuery message type (the type of message related to inline_keyboard) an endQuery response is sent back to Telegram as feedback.

joroMaser, 2022-02-08:
2- strlen is compute in compile time so if you put name for "12" it will be more clean code 
5- Can you tell me more about this software watchdog about esp8266 ? I got this software watchdog in [this library](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot) , so I hope that relate to that library is not async....

about Inline_keyboard 

1)
I see:
`myBot.endQuery(msg, "Light on", true);`
But it is not case sensitive ? The [callback ](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/keyboards/keyboards.ino#L59) is 
`#define LIGHT_ON_CALLBACK  "lightON"`

2)
There is no validation when sending message ,  I can send
String keyboardJson = "{\"inline_keyboard\": [{ \"text\" : \" AAAA \", \"callback_data\" : \"BBBB\" }]}";
and Telegram will response:
`getUpdates():	{"ok":false,"error_code":400,"description":"Bad Request: field \"inline_keyboard\" of the InlineKeyboardMarkup must be an Array of Arrays"}`
Maybe it is right to add some validation before sending?

3)I think there is an error while parse MessageQuery message

I send Inline_keyboard like :
      String keyboardJson = "{\"inline_keyboard\": [[{ \"text\" : \" press here!\", \"callback_data\" : \"press1\" }]]}";
      myBot.sendMessage(msg, "Hi", keyboardJson);

and when user press on **press here** I got this json:

```
{
  "ok": true,
  "result": [
    {
      "update_id": XXXXXXXXXXX,
      "callback_query": {
        "id": "XXXXXXXXXXX",
        "from": {
          "id": XXXXXXXXXXX,
          "is_bot": false,
          "first_name": "XXXXXXXXXXX",
          "language_code": "he"
        },
        "message": {
          "message_id": 1130,
          "from": {
            "id": XXXXXXXXXXX,
            "is_bot": true,
            "first_name": "XXXXXXXXXXX",
            "username": "XXXXXXXXXXX"
          },
          "chat": {
            "id": XXXXXXXXXXX,
            "first_name": "XXXXXXXXXXX",
            "type": "private"
          },
          "date": XXXXXXXXXXX,
          "text": "Hi",
          "reply_markup": {
            "inline_keyboard": [
              [
                {
                  "text": " press here!",
                  "callback_data": "press1"
                }
              ]
            ]
          }
        },
        "chat_instance": "XXXXXXXXXXX",
        "data": "press1"
      }
    }
  ]
}
```
But when I tried to print msg.callbackQueryData I got garbage ....

```
    if (msg.messageType == MessageQuery)
    {
      Serial.println(msg.callbackQueryData );

    }
```
That print

```
XXXXXXXXXXX/sendMessage HTTP/1.0
Host: api.telegram.org
Connection: keep-alive
Content-Type: application/json
Content-Length: 153

{"chat_id":xxxxxxx,"text":"Hi","parse_mode":"HTML","reply_markup":{"inline_keyboard":[[{"text":" press here!","callback_data":"press1"}]]}}
```

cotestatnt, 2022-02-09:
Regarding strlen() you are right, it's clearer.
Soft WD happens more or less once a day and every time when library is trying to re-establish connection to Telegram server. The SSL handshake gets timeout and then ESP restart due to watchdog.
At the moment, I'm testing with `client.setInsecure()` option enabled since this morning.

Inline_keyboard:

The text you send back with `endQuery()` it's not related to the text associated to the inline button, but it will be displayed to the user as a notification. [Check here for more details](https://core.telegram.org/bots/api#answercallbackquery).
Validation would be nice feature, but it's very heavy to implement and actually I haven't enough time at the moment :(  We will see in the future.

``` 
if (msg.callbackQueryData == "press1")
```
You can't compare a const char* with the == operator, you should use `strcmp()` or store `msg.callbackQueryData` to a local String in order to be able to use == and the other methods of class String.

For the garbage output, I was not able to reproduce the issue. Can you share the whole test code?

joroMaser, 2022-02-09:
I think there is a  problem about parsing MessageQuery
I add 1 more member to `struct TBMessage` 
`String         callbackQueryData1;`
And then I add 1 more line for parsing `MessageQuery` 

`message.callbackQueryData1 = updateDoc["result"][0]["callback_query"]["data"].as<String>();`

Now I tried to print both `callbackQueryData1 ` and `callbackQueryData` 
callbackQueryData1  print **press1** but callbackQueryData print garbage as I mentioned before .

```
void loop() {

  TBMessage msg;

  if (myBot.getNewMessage(msg)) {

    String keyboardJson = "{\"inline_keyboard\": [[{ \"text\" : \" press here!\", \"callback_data\" : \"press1\" }]]}";
    myBot.sendMessage(msg, "Hi", keyboardJson);

    if (msg.messageType == MessageQuery)
    { 
      Serial.println("++++++++++++++++++++++");
      Serial.println(msg.callbackQueryData );
      Serial.println("++++++++++++++++++++++");
      Serial.println(msg.callbackQueryData1 );
      Serial.println("++++++++++++++++++++++");

      myBot.endQuery(msg, "OK", true);

    }
  }

}
```

joroMaser, 2022-02-09:
I think there is a  problem about parsing char * in esp8266 from JSON 

For example in `MessageQuery`
I add 1 more member to `struct TBMessage` 
`String         callbackQueryData1;`
And then I add 1 more line for parsing `MessageQuery` 

`message.callbackQueryData1 = updateDoc["result"][0]["callback_query"]["data"].as<String>();`

Now I tried to print both `callbackQueryData1 ` and `callbackQueryData` 
callbackQueryData1  print **press1** but callbackQueryData print garbage as I mentioned before .

```
void loop() {

  TBMessage msg;

  if (myBot.getNewMessage(msg)) {

    String keyboardJson = "{\"inline_keyboard\": [[{ \"text\" : \" press here!\", \"callback_data\" : \"press1\" }]]}";
    myBot.sendMessage(msg, "Hi", keyboardJson);

    if (msg.messageType == MessageQuery)
    { 
      Serial.println("++++++++++++++++++++++");
      Serial.println(msg.callbackQueryData );
      Serial.println("++++++++++++++++++++++");
      Serial.println(msg.callbackQueryData1 );
      Serial.println("++++++++++++++++++++++");

      myBot.endQuery(msg, "OK", true);

    }
  }

}
```

1 more example in `MessageText` :  I jot that JSON

```
{
  "ok": true,
  "result": [
    {
      "update_id": XXXXXXXX,
      "message": {
        "message_id": 123,
        "from": {
          "id": XXXXXXXX,
          "is_bot": false,
          "first_name": "XXXXXXXXXX",
          "language_code": "en"
        },
        "chat": {
          "id": -XXXXXXXX,
          "title": "XXXXXXXX",
          "type": "supergroup"
        },
        "date": XXXXXXXX,
        "text": "Gggh"
      }
    }
  ]
}

```
But when I tried to print `msg.sender.firstName` I got empty string and `strlen(msg.sender.firstName)` print 1

But when I add
`String        firstName1;` to  `struct TBUser` and then parse it 
`message.sender.firstName1 = updateDoc["result"][0]["message"]["from"]["first_name"].as<String>();;`
I can print the first name as well 

So maybe there is a problem with parsing to char * ? can you please check it ? maybe we need to convert all char * to String in all structs ?
**I think that `updateDoc` holds array of String , not array of char***

By the way even I use
`myBot.endQuery(msg, "OK", true);` that didn't stop the loading cycle even the function return true(1)
maybe because when I connect to my bot I see ` "supports_inline_queries": false` ?
```

getMe():	
{
  "ok": true,
  "result": {
    "id": XXXXX,
    "is_bot": true,
    "first_name": XXXXXXX",
    "username": "XXXXXXXXX",
    "can_join_groups": true,
    "can_read_all_group_messages": false,
    "supports_inline_queries": false
  }
}

```

closed 2022-02-24 by cotestatnt