cotestatnt/AsyncTelegram2 #51: How to disable the bot?

phazz1980, 2022-02-13:
When for some reason the connection to the bot router is lost when trying to connect, the ESP 32 cycle becomes 20 seconds. How can I turn it off when there is no connection?

P.S. I don’t know English well, translated through Google )

phazz1980, 2022-02-13:
00:52:28.002 -> [D][AsyncTelegram2.cpp:28] checkConnection():	Start handshaking...
this freeze

cotestatnt, 2022-02-24:
Hi @phazz1980 
I'm sorry for late reply, but i'm very busy in these days.

You could check the connection in the main loop() and avoid to call ` if (myBot.getNewMessage(msg)) {  ... ` which is then 
 responsible to call checkConnection() method.

cotestatnt, 2022-03-06:
I'm going to close this issue for inactivity.
Reopen if you need more info.

closed 2022-03-06 by cotestatnt

phazz1980, 2022-03-10:
This situation occurs when there is a wi-fi connection but no internet connection

phazz1980, 2022-03-10:
if (initialized && WiFi.isConnected() &&myBot.getNewMessage(msg))