cotestatnt/AsyncTelegram2 #52: Sending messages

gravedigger21, 2022-02-17:
Hi!
I continue to deal with your library and faced the fact that I can not send a message to the telegram when any events occur. For example, write the following code:
`void loop()
{
TBMessage msg;
if(digitalRead(button)==HIGH) myBot.sendMessage(msg, " button=1");
}`
but nothing comes in telegram. What am I doing wrong?

cotestatnt, 2022-02-24:
Hi @gravedigger21 
I'm sorry for late reply, but i'm very busy in these days.

In the examples included in library, the bot send a message to user as a reply, so the `struct TBMessage msg` is already valorized with all necessary data.

In order to be able to send directly a message to user (or to groups), you need to know the ID in advance.
For example in [echoBot.ino](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/echoBot/echoBot.ino#L96) you can find these lines of code:

 ```
  // Send a message to specific user who has started your bot
  // Target user can find it's own userid with the bot @JsonDumpBot
  // https://t.me/JsonDumpBot
  int64_t userid = 123456789;  
  myBot.sendTo(userid, welcome_msg);
```

With your sample code it will be:

```
// https://t.me/JsonDumpBot
int64_t userid = 123456789;  
.....
void loop() { 
  if(digitalRead(button)==HIGH) {    
    myBot.sendTo(userid, " button=1"); 
  }
}
```

You could use also the sendMessage() method, but fitting the msg struct with the ID, but I think sendTo() is more "clear":
```
void loop() { 
  if(digitalRead(button)==HIGH) {
     TBMessage msg;
     msg.chatId = 123456789;
     myBot.sendMessage(msg, " button=1"); 
  }
}
```

cotestatnt, 2022-03-06:
I'm going to close this issue for inactivity.
Reopen if you need more info.

closed 2022-03-06 by cotestatnt