cotestatnt/AsyncTelegram2 #54: Error replaying a message

kwecko, 2022-02-24:
When replaying a message that the bot sent, it stops working, not responding to any more messages.

Looking at the serial, the following information is printed:

{"ok":true,"result":[{"update_id":289757217,
"message":{"message_id":50,"from":{"id":251360319,"is_bot":false,"first_name":"Kwecko","username":"kwecko","language_code":"pt-br"},"chat":{"id":251360319,"first_name":"Kwecko","username":"kwecko","type":"private"},"date":1645725307,"reply_to_message":{"message_id":47,"from":{"id":5254501027,"is_bot":true,"first_name":"Kdesenvbot","username":"kdesenvbot"},"chat":{"id":251360319,"first_name":"Kwecko","username":"kwecko","type":"private"},"date":1645725268,"text":"Os comandos dispon\u00edveis s\u00e3o:\n\n /temperatura : Para verificar a temperatura\n /humidade : Para verificar a humidade\n /pressao : Para verificar a press\u00e3o\n /gas : Para verificar o g\u00e1s\n /mac : Para verificar o endere\u00e7o MAC\n /ip : Para verificar o endere\u00e7o IP do ESP32\n /timeboot : Tempo em que o ESP est\u00e1 ligado\n /version : Informa a vers\u00e3o do Sistema\n /esptemp : Informa a temperatura do ESP\n /help : Listar os comandos dispon\u00edveis","entities":[{"offset":31,"length":12,"type":"bot_command"},{"offset":31,"length":12,"type":"bold"},{"offset":76,"length":9,"type":"bot_command"},{"offset":76,"length":9,"type":"bold"},{"offset":115,"length":8,"type":"bot_command"},{"offset":115,"length":8,"type":"bold"},{"offset":152,"length":4,"type":"bot_command"},{"offset":152,"length":4,"type":"bold"},{"offset":181,"length":4,"type":"bot_command"},{"offset":181,"length":4,"type":"bold"},{"offset":219,"length":3,"type":"bot_command"},{"offset":219,"length":3,"type":"bold"},{"offset":264,"length":9,"type":"bot_command"},{"offset":264,"length":9,"type":"bold"},{"offset":308,"length":8,"type":"bot_command"},{"offset":308,"length":8,"type":"bold"},{"offset":348,"length":8,"type":"bot_command"},{"offset":348,"length":8,"type":"bold"},{"offset":389,"length":5,"type":"bot_command"},{"offset":389,"length":5,"type":"bold"}]},"text":"/help","entities":[{"offset":0,"length":5,"type":"bot_command"}]}}]}

this message is printed constantly. Does anyone have any suggestions on how to get around this problem?

Thanks

cotestatnt, 2022-02-27:
Hi @kwecko 
It seeems that your JSON message from Telegram is very long due to the entities embedded (u00e3o, u00e7o etc etc)

I have not found the way to avoid the list of entities (useless for us) at the end of the message and with such a JSON the [ArduinoJSON ](https://arduinojson.org/v6/assistant/)library fails the deserialization.

You would need to increase the buffer up to at least 3072 bytes editing this https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/DataStructures.h#L5.

A simpler solution could be break the long message into 2/3 parts.

cotestatnt, 2022-03-06:
I'm going to close this issue for inactivity.
Reopen if you need more info.

closed 2022-03-06 by cotestatnt

kwecko, 2022-03-06:
Sorry Cotestatnt for responding late. I did what you suggested and dividing the message into 3 parts. With that, the problem was solved. I didn't want to change the buffet value for fear that a future update might make the problem appear again.
Thank you for your support.