cotestatnt/AsyncTelegram2 #56: Not working with ESP32 Arduino SDK 2.0.2

milGus, 2022-03-04:
Works with 2.0.1. However with 2.0.2, we get "Unable to connect to Telegram server"

milGus, 2022-03-04:
Verbose debug

[ 12290][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[ 12296][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[ 12303][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 217580
[ 12311][V][ssl_client.cpp:67] start_ssl_client(): Starting socket
[ 13079][V][ssl_client.cpp:143] start_ssl_client(): Seeding the random number generator
[ 13081][V][ssl_client.cpp:152] start_ssl_client(): Setting up the SSL/TLS structure...
[ 13084][V][ssl_client.cpp:175] start_ssl_client(): Loading CA cert
[ 13138][V][ssl_client.cpp:244] start_ssl_client(): Setting hostname for TLS session...
[ 13139][V][ssl_client.cpp:259] start_ssl_client(): Performing the SSL/TLS handshake...
[ 13146][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1
[ 13150][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.

Unable to connect to Telegram server

milGus, 2022-03-04:
Update: https://github.com/espressif/arduino-esp32/issues/6077

It appears its a regression in 2.0.2

Workaround:

client.stop();
client.setHandshakeTimeout(30);

Now it works OK.

cotestatnt, 2022-03-05:
Thanks @milGus 
I'm sorry for late reply.

I've implemented the workaround suggested directly in the library inside the checkConnection() method and it works!
It will be avalible online soon.

```
// ESP32 core version 2.0.2 workaround
// https://github.com/espressif/arduino-esp32/issues/6077
#if defined(ESP32)
WiFiClientSecure* _client =  static_cast<WiFiClientSecure*>(telegramClient);
_client->setHandshakeTimeout(30);
#endif
```

milGus, 2022-03-05:
Thanks always happy to help. Btw is there something I can do to help with the GSM/LTE integration?

cotestatnt, 2022-03-06:
> is there something I can do to help with the GSM/LTE integration?

A working example would be very very apreciated! 
A user wrote that he was able to work via GPRS and this [SSL wrapper](https://github.com/govorox/SSLClient) with a SIM800 if I remember correctly, but I haven't had any updates from him.

closed 2022-04-23 by cotestatnt