cotestatnt/AsyncTelegram2 #58: Get message from group by message id

joroMaser, 2022-03-13:
How can I get only message text by message ID in group, Even that message is not new message?

cotestatnt, 2022-03-16:
Actually there'arent methods that directly do what you need, but maybe [forwardMessage()](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.h#L162) or [editMessage()](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.h#L347) can help you for your purpose?

joroMaser, 2022-03-16:
I want to use telegram as database.
So I want to save 1 message in group so I can update this message (from my phone) and each time esp will wake up it will read that message
@cotestatnt

cotestatnt, 2022-03-16:
I've done some tests with forwardMessage()... 
It's a little bit difficult because in the reply from server WE HAVE the text of specific message, but formatted with a little different JSON.

In particular, with a "normal" message the reply is something like this:
```
{ "ok": true, "result":  [{ ... }]}
```
So in order to parse you have to use doc["result"][0] to get the first element of JSON array.

When you forward a message, the reply is without array, and this requires to review all the JSON parsing function.
```
{ "ok": true, "result":  { ... }}
```
As soon as I have some time I try to see if there is a smart way to do it

closed 2022-03-22 by joroMaser