cotestatnt/AsyncTelegram2 #59: Send photo to group!

javierferwolf, 2022-03-15:
hello everyone and thank you very much for this library that is very useful!  I know this question has been asked before but I can't send a photo to a group I have tried to send as follows

int64_t userid = -100xxxxxxx;
if (msgText.equalsIgnoreCase("/picfs3")) {
     File file = FILESYSTEM.open("/pic.jpg", "r");
     myBot.sendPhotoByFile(userid, file, file.size());
     file.close();
}
     
but i have an error:

C:\Users\Javierlupus\Documents\Arduino\nuevablbliotecaprimera_version\nuevablbliotecaprimera_version.ino: In function 'void loop()':
nuevablbliotecaprimera_version:326:58: error: no matching function for call to 'AsyncTelegram2::sendPhotoByFile(int64_t&, fs::File&, size_t)'
  326 |           myBot.sendPhotoByFile(userid, file, file.size());
      |                                                          ^
In file included from C:\Users\Javierlupus\Documents\Arduino\nuevablbliotecaprimera_version\nuevablbliotecaprimera_version.ino:1:
C:\Users\Javierlupus\Documents\Arduino\libraries\AsyncTelegram2-master\src/AsyncTelegram2.h:253:17: note: candidate: 'bool AsyncTelegram2::sendPhotoByFile(int64_t, Stream*, size_t)'
  253 |     inline bool sendPhotoByFile(int64_t chat_id, Stream *stream, size_t size) {
      |                 ^~~~~~~~~~~~~~~
C:\Users\Javierlupus\Documents\Arduino\libraries\AsyncTelegram2-master\src/AsyncTelegram2.h:253:58: note:   no known conversion for argument 2 from 'fs::File' to 'Stream*'
  253 |     inline bool sendPhotoByFile(int64_t chat_id, Stream *stream, size_t size) {
      |                                                  ~~~~~~~~^~~~~~
C:\Users\Javierlupus\Documents\Arduino\libraries\AsyncTelegram2-master\src/AsyncTelegram2.h:258:17: note: candidate: 'bool AsyncTelegram2::sendPhotoByFile(int64_t, const char*, fs::FS&)'
  258 |     inline bool sendPhotoByFile(int64_t chat_id, const char* filename, fs::FS &fs) {
      |                 ^~~~~~~~~~~~~~~
C:\Users\Javierlupus\Documents\Arduino\libraries\AsyncTelegram2-master\src/AsyncTelegram2.h:258:62: note:   no known conversion for argument 2 from 'fs::File' to 'const char*'
  258 |     inline bool sendPhotoByFile(int64_t chat_id, const char* filename, fs::FS &fs) {
      |                                                  ~~~~~~~~~~~~^~~~~~~~
exit status 1
no matching function for call to 'AsyncTelegram2::sendPhotoByFile(int64_t&, fs::File&, size_t)'

any help please?

cotestatnt, 2022-03-15:
Hi Javier,
`sendPhotoByFile()` method is for backward compatibility and sooner or later will be removed in the next releases.
It's better if you use this method

```
bool sendPhoto(int64_t chat_id, uint8_t *data, size_t size) 
```

javierferwolf, 2022-03-16:
Hi @cotestatnt, thanks for your answer! 

And yes in used this method  myBot.(userid, file, file.size());  but it doesn't work, it just doesn't send the photo to the group, only to private chat

Any suggestions of what could be going on?

thank you

cotestatnt, 2022-03-16:
Thanks @javierferwolf for finding this annoying bug.
The problem is due to the fact that the group id is a negative int64 and the convert to string function was not handling such number correctly.

I changed the [AsyncTelegram2.cpp](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.cpp) source.
If you don't want wait for the next release and try the changes right away, you can directly overwrite the file with the new version.

In addition, when you send a message from a group, Telegram add automatically the postfix @username after the message text, so you have to handle properly inside the sketch.

javierferwolf, 2022-03-17:
@cotestatnt Thank you very much for your answer.
and well honestly I will wait for a new version from you =)  since unfortunately I do not have much experience to change for a new version. I just don't know how to change or in which part of the code [AsyncTelegram2.cpp](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.cpp) to make the change
thank you

javierferwolf, 2022-03-24:
@cotestatnt  I know I should post another post, but I have also tried to call the keyboard to the group "myReplyKbd" but this also doesn't work only the keyboard "myInlineKbd" works.  
Will there be any fix also in the new update?

cotestatnt, 2022-03-25:
Hi @javierferwolf.

If you are using the provided **keyboard.ino** example as starting point, you need to know that some data can't be send in a group due to the privacy restrictions like for example location and user contacts so you need to remove this from **myReplyKbd**

For the future, when you found such a similar behaviuor, check the debug messages from library [setting this line to 1](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.h#L25)
For example, this is the message recevied from Telegram server when I've tried to send message with example as is: it's clear the response and the reason it doesn't work.

```C++
[E][AsyncTelegram2.cpp:176] getUpdates(): {"ok":false,"error_code":400,"description":"Bad Request: location can be requested in private chats only"}
```

closed 2022-04-23 by cotestatnt