cotestatnt/AsyncTelegram2 #61: inline_keyboard didn't send if message without text

joroMaser, 2022-03-19:
```
  String keyboardJson = "{\"inline_keyboard\": [[{ \"text\" : \" TEST\", \"callback_data\" : \"TTTTTT\" }]]}";
  myBot.sendMessage(msg, "  ", keyboardJson);
```
That message will not send, Why is that?

cotestatnt, 2022-03-21:
This is a limit from Telegram and not due to library.
A message cannot be empty and even if you sent blank spaces it will be recognized as empty.

If you enable library debug printing, you can see the server response is as follows:

`{"ok":false,"error_code":400,"description":"Bad Request: message must be non-empty"}`

closed 2022-03-21 by cotestatnt

joroMaser, 2022-03-27:
@cotestatnt the problem is the `sendMessage` always return 0 . When message successfully sent and when message failed like in my example