cotestatnt/AsyncTelegram2 #62: ESP32 on keyboards.ino keep on rebooting in a loop

Dcmwrnd, 2022-03-21:
Hi Owner,

I added a “*reboot*” message at line 155 to restart the EPS32 when necessary. But after the first *reboot* message send, the ESP32 keep on rebooting in a loop. Every ESP32 restart, Telegram will sent out the last message again & the ESP32 keeps rebooting.

Can I clear the last message send to Telegram?

Compile with:
Arduino IDE:     v1.8.19
ArduinoJson:     v6.19.3
Board Manager:   v1.0.6
AsyncTelegram2:  v2.1.0

/*
  Name:        keyboards.ino
  Created:     05/03/2022
  Author:      Tolentino Cotesta <cotestatnt@yahoo.com>
  Description: a more complex example that do:
             1) if a "/inline_keyboard" text message is received, show the inline custom keyboard,
                if a "/reply_keyboard" text message is received, show the reply custom keyboard,
                otherwise reply the sender with "Try /reply_keyboard or /inline_keyboard" message
             2) if "LIGHT ON" inline keyboard button is pressed turn on the LED and show a message
             3) if "LIGHT OFF" inline keyboard button is pressed, turn off the LED and show a message
             4) if "GitHub" inline keyboard button is pressed,
                open a browser window with URL "https://github.com/cotestatnt/AsyncTelegram"
*/

/*
  Compile with versions:
  Arduino IDE:     v1.8.19
  ArduinoJson:     v6.19.3
  Board Manager:   v1.0.6
  AsyncTelegram2:  v2.1.0
*/

#include <AsyncTelegram2.h>

// Timezone definition
#include <time.h>
#define MYTZ "CET-1CEST,M3.5.0,M10.5.0/3"
//#define MYTZ "SGT-8"

/*
  Set true if you want use external library for SSL connection 
  For example https://github.com/OPEnSLab-OSU/SSLClient/ is very efficient BearSSL library and 
  you can use AsyncTelegram2 even with other MCUs or transport layer (ex. Ethernet or GPRS)
*/
#define USE_CLIENTSSL false

#include <WiFi.h>
#include <WiFiClient.h>

#if USE_CLIENTSSL
    #include <SSLClient.h>      // https://github.com/OPEnSLab-OSU/SSLClient/
    #include "tg_certificate.h"
    WiFiClient base_client;
    SSLClient client(base_client, TAs, (size_t)TAs_NUM, A0, 1, SSLClient::SSL_ERROR);
  #else
    #include <WiFiClientSecure.h>
    WiFiClientSecure client;
#endif

AsyncTelegram2 myBot(client);

const char* ssid  =  "xxxxxxxxx";     // SSID WiFi network
const char* pass  =  "xxxxxxxxx";     // Password  WiFi network
const char* token =  "xxxxxxxxxxxx";  // Telegram token

// Check the userid with the help of bot @JsonDumpBot or @getidsbot (work also with groups)
// https://t.me/JsonDumpBot  or  https://t.me/getidsbot

[... 169 lines cut ...]
          reply += "; Latitude: ";
          reply += msg.location.latitude;
          Serial.println(reply);
          myBot.sendMessage(msg, reply);
          break;
        }

      case MessageContact: {
          // received a contact message
          String reply = "Contact information received:";
          reply += msg.contact.firstName;
          reply += " ";
          reply += msg.contact.lastName;
          reply += ", mobile ";
          reply += msg.contact.phoneNumber;
          Serial.println(reply);
          myBot.sendMessage(msg, reply);
          break;
        }

      default:
        break;
    }
  }
}

cotestatnt, 2022-03-21:
In order to handle properly reboot from Telegram command, you have to wait that message is fully  handled before doing restart otherwise the bot read again the last message (restart command) and reboot again and so on in an endless loop.

Check this lines in [this example](https://github.com/cotestatnt/AsyncTelegram2/blob/0104e4d832ca5058c9f652b6face593becd62b0f/examples/ESP8266/OTA_password/OTA_password.ino#L192)

Dcmwrnd, 2022-03-22:
Thank you Sir, it is working perfectly now after inserting the noNewMessage()

closed 2022-03-22 by cotestatnt