cotestatnt/AsyncTelegram2 #66: Struct TBGroup removed!

kwecko, 2022-03-24:
I updated the library to version 2.1.1 and noticed that the Struct TBGroup was removed. I use the struct for read the ID of the group to  that to determine group can send message to the system.  So, now, How I can check the ID of message and to determine if is message valid or no, based of the ID the Group ?

Thanks!

cotestatnt, 2022-03-25:
Hi @kwecko 
I'm sorry for the inconvenience, but I've changed a little the data organization because a group is just a chat with the ID negative, so keeping track with a specialized struct I think it's just a waste of memory most of time and the code for sending messages it's simpler an clearer if I don't have to distinguish groups from users.

You can found information about group like for a "normal" user, so if you need the ID it will be stored always in `msg.chatId`
If you need information about member of group who sent message, you will find in `msg.sender`
And if you need infos about members entering or leaving the groups you will find in `msg.member`

If you enable the debug messages you can have a clear view of all nformations sent from Telegram belong a new message.
For example this is the JSON received with a message sent to my bot from a group:

```JSON
{
  "update_id": 764254158,
  "message": {
    "message_id": 30783,
    "from": {
      "id": 123456789,                       // This is the ID of user who send message
      "is_bot": false,
      "first_name": "myFirstname",
      "username": "myUsername",
      "language_code": "it"
    },
    "chat": {
      "id": -1234567890,                     // This is the ID of chat: it's a negative int64 for groups
      "title": "myGroupTitle",
      "type": "group",
      "all_members_are_administrators": true
    },
    "date": 1648194423,
    "text": "ciao ciao"
  }
}
```

kwecko, 2022-03-25:
Thanks my friend.
I did what you suggested and analyzed the msg.chatId and achieved resolve my problem.

Thanks by you attention .

cotestatnt, 2022-03-25:
You're welcome!
Thank you for using this library.

closed 2022-03-25 by cotestatnt