cotestatnt/AsyncTelegram2 #72: help solve the problem. resending messages

Demon-MC, 2022-04-24:
When sending the help command to telegram, esp8266 sends the "Help" message without stopping

`    else if (msgText.equals("Помощь")){
      String help = "Ниже приведен список всех команд.\n";
      help += "Отопвыкл, Отопвкл - Выключение и включение отопителя, по умолчанию отопитель выключен.\n";
       help += "Макс18, Макс20, Макс25 - Максимальная температура для поддержания отопителем.\n";
       help += "Печьвыкл, Печьвкл - Выключение и включение печки принудительно.\n";
      help += "Охранавкл, Охранавыкл - Включает обнаружение движения, по умолчанию охрана выключена.\n";
      help += "Инфо - Предоставление информации о всех датчиках системы умная дача";  
     myBot.sendMessage(msg, help);
    }`

everything worked fine in AsyncTelegram1. I found a solution to the problem: if you reduce the length of the message in String, one message is sent to the chat..

how to make it so that long messages are sent to the chat without endless repetitions

cotestatnt, 2022-04-24:
Hi @Demon-MC 
The problem is that Cyrillic characters that are treated as "entities" and then your message becomes a very long list of html entities like this:

`"text":"\u041d\u0438\u0436\u0435 \u043f\u0440\u0438\u0432\u0435\u0434\u0435\u043d \u0441\u043f\u0438\u0441\u043e\u043a \u0432\u0441\u0435\u0445 \u043a\u043e\u043c\u0430\u043d\u0434.\n\u041e\u0442\u043e\u043f\u0432\u044b\u043a\u043b, \u041e\u0442\u043e\u043f\u0432\u043a\u043b - \u0412\u044b\u043a\u043b\u044e\u0447\u0435\u043d\u0438\u0435 \u0438 \u0432\u043a\u043b\u044e\u0447\u0435\u043d\u0438\u0435 \u043e\u0442\u043e\u043f\u0438\u0442\u0435\u043b\u044f, \u043f\u043e \u0443\u043c\u043e\u043b\u0447\u0430\u043d\u0438\u044e \u043e\u0442\u043e\u043f\u0438\u0442\u0435\u043b\u044c \u0432\u044b\u043a\u043b\u044e\u0447\u0435\u043d.\n\u041c\u0430\u043a\u044118, \u041c\u0430\u043a\u044120, \u041c\u0430\u043a\u044125 - \u041c\u0430\u043a\u0441\u0438\u043c\u0430\u043b\u044c\u043d\u0430\u044f \u0442\u0435\u043c\u043f\u0435\u0440\u0430\u0442\u0443\u0440\u0430 \u0434\u043b\u044f \u043f\u043e\u0434\u0434\u0435\u0440\u0436\u0430\u043d\u0438\u044f \u043e\u0442\u043e\u043f\u0438\u0442\u0435\u043b\u0435\u043c.\n\u041f\u0435\u0447\u044c\u0432\u044b\u043a\u043b, \u041f\u0435\u0447\u044c\u0432\u043a\u043b - \u0412\u044b\u043a\u043b\u044e\u0447\u0435\u043d\u0438\u0435 \u0438 \u0432\u043a\u043b\u044e\u0447\u0435\u043d\u0438\u0435 \u043f\u0435\u0447\u043a\u0438 \u043f\u0440\u0438\u043d\u0443\u0434\u0438\u0442\u0435\u043b\u044c\u043d\u043e.\n\u041e\u0445\u0440\u0430\u043d\u0430\u0432\u043a\u043b, \u041e\u0445\u0440\u0430\u043d\u0430\u0432\u044b\u043a\u043b - \u0412\u043a\u043b\u044e\u0447\u0430\u0435\u0442 \u043e\u0431\u043d\u0430\u0440\u0443\u0436\u0435\u043d\u0438\u0435 \u0434\u0432\u0438\u0436\u0435\u043d\u0438\u044f, \u043f\u043e \u0443\u043c\u043e\u043b\u0447\u0430\u043d\u0438\u044e \u043e\u0445\u0440\u0430\u043d\u0430 \u0432\u044b\u043a\u043b\u044e\u0447\u0435\u043d\u0430.\n\u0418\u043d\u0444\u043e - \u041f\u0440\u0435\u0434\u043e\u0441\u0442\u0430\u0432\u043b\u0435\u043d\u0438\u0435 \u0438\u043d\u0444\u043e\u0440\u043c\u0430\u0446\u0438\u0438 \u043e \u0432\u0441\u0435\u0445 \u0434\u0430\u0442\u0447\u0438\u043a\u0430\u0445 \u0441\u0438\u0441\u0442\u0435\u043c\u044b \u0443\u043c\u043d\u0430\u044f \u0434\u0430\u0447\u0430"`

In these cases, unfortunately the response from the server is splitted into packets and then parsing of the received message is done when it is not yet complete.
I have pinpointed exactly where this happens, but I need to do some testing to find the best way to avoid this problem.

Demon-MC, 2022-04-24:
the most interesting thing is that the message is sent in full

Demon-MC, 2022-04-24:
I have solved the problem so far, this is a temporary option

unsigned long Timehelp = 0;  //хранение времени отправка повторных сообщений
else if (msgText.equals("Помощь")){
         
    if(millis() - Timehelp > 10000){
     String help = "Ниже приведен список всех команд.\n";
      help += "Отопвыкл, Отопвкл - Выключение и включение отопителя, по умолчанию отопитель выключен.\n";
       help += "Макс18, Макс20, Макс25 - Максимальная температура для поддержания отопителем.\n";
       help += "Печьвыкл, Печьвкл - Выключение и включение печки принудительно.\n";
      help += "Охранавкл, Охранавыкл - Включает обнаружение движения, по умолчанию охрана выключена.\n";
      help += "Инфо - Предоставление информации о всех датчиках системы умная дача";  
      myBot.sendMessage(msg, help); 
      Timehelp = millis();} 
    
    }

Demon-MC, 2022-04-24:
a repeat message can only be sent after 10 seconds

cotestatnt, 2022-04-25:
Hi @Demon-MC 
I've just uploaded a fix, check with latest release please.

closed 2022-04-29 by cotestatnt