cotestatnt/AsyncTelegram2 #74: msg.sender.username  is "null" ?

konig87nikkinakki, 2022-04-27:
Hi, maybe it's not a issue: probably it's me a beginner.
I wrote you about 8 months ago and you answered me kindly, i m sad to bore you again.
Well, i m trying to run LIGHT BOT.ino  to turn on and off a led. when i write to bot, it answers to me "welcome null. Try /light_on or /light off".
1) my first question is why msg.sender.username is null?
2) my second question is: how can i print on Serial.println or to a Telegram message the Sender ID ? everyone has got one different, i would like to receive something as: "welcome [12345688809], try these commands" 
with my past libraries (not good working, that's why i want to change to yours) it was msg.sender.id .. i m a beginner and i dont know how to discover commands or strings into libraries, can you please give me some help?
Thank you so much

if could help, i have installed the last json library and your last library, if you need i write you exact versions. i m using nodemcu esp8266

cotestatnt, 2022-04-28:
Hi @konig87nikkinakki 
If you give me some additional info it's better because I've just tried [lightBot.ino](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/lightBot/lightBot.ino) example with a NodeMCU 0.9 and I can get as expected both username and id.

konig87nikkinakki, 2022-04-29:
Well! thank you for reply and sorry for late answer. i have tested with esp8266 nodemcu and esp32 . my situation is:
-> Arduino ide    1.8.12
-> AsyncTelegram2    2.1.2
-> Arduino Json    6.19.4
-> esp32 v 1.0.6

i compiled LIGHT BOT. INO in esp32 and in nodemcu. in serial monitor i see:
**_Test Telegram connection... OK
Bot name: @ciaoninniciao_bot_**

and this is ok.

if i send message from my smartphone to my bot it works and anwers but if i write a wrong command as "HELLOOO" it answers to my smartphone:
_**Welcome null.
Try /light_on or /light_off**_

i need to write my ID, that is the number that changes in every smartphone. mine is (i obtained writing to a bot) 755575XXX

The bot i m using is a test bot that i created with botfather, 
1989143778:AAGx67iAR_o54zyTYQIwYEMHZ73Cx1_z[xxx]

Thank you so much.
my esp32 is EZ DELIVERY ESP32-WROOM-32.
any help?

konig87nikkinakki, 2022-04-29:
hey! i just modified lightbot.ino  last lines as following:

```
    else {                                              // otherwise...
      // generate the message for the sender
      String reply;
      reply = "Welcome " ;
      reply += String(long(msg.sender.id)); 
      reply += "\n";      
      reply += String(float(msg.location.latitude));   
      reply += "\n";         
      reply += String(float(msg.location.longitude));        
      reply += ".\nTry /light_on or /light_off ";
      myBot.sendMessage(msg, reply);                    // and send it
    }
```

and it returns me my ID. Well, it works. but it returns to me **-0.00 and 0.00 as Latitude and Longitude**. i ve turn on my GPS and allowed telegram to read position on my smartphone, but it is shy and doesnt work 
Any suggestions?

cotestatnt, 2022-04-30:
<quote>-> esp32 v 1.0.6</quote>

There is a particular reason why you are using a such old version of ESP32 Arduino core?
I would suggest you to upgrade to version 2.0.1 (with latest 2.0.2 I've encountered some problems with HTTPS connections).

Regarding latitude and longitude, the value is valid only if you send a location type of message and not with every message you send to bot (this is not allowed for privacy reason, you must send location with explicity command).

Check the example [keyboards.ino](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/keyboards/keyboards.ino) where a custom keyboard with some buttons is created and one of the button is the request for location.

One last point: with String operator **+** or **+=** it's not necessary using the constructors `String()` or `long()` or `float()`. 
It's just a waste of SRAM because you are useless duplicating the values.

The problem is that the String class does not support the `int64_t` type, so forcing cast to `long` can lead with a resulting value completely wrong (long == int32_t).
Even if your ID is a number fewer than maximum possible with `int32_t` variable, it's better use an array of char and the `snprintf()` statement instead:

```C++
 char reply[100];
 snprintf(reply, sizeof(reply), "Welcome %lld.\nTry /light_on or /light_off",  msg.sender.id);
 myBot.sendMessage(msg, reply);    
```

konig87nikkinakki, 2022-04-30:
**-> esp32 v 1.0.6**

That s because it's the higher version i can see in the list
[image]

maybe because i dont have last arduino ide installed? i usually don t want to upgrade until things work, because some updates may lose old commands or features. i use other libraries for other shields (display, temp sensors, accelerometers) so usually i dont update until necessary. i can see your library works very very good

cotestatnt, 2022-04-30:
No, you can't see the latest version because the wrong json link in the preferences.
The updated is 
`https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json`

[image]

konig87nikkinakki, 2022-05-06:
Sorry for late answer. Thank you, i updated ESP32 as you wrote and now it works!
i had https://dl.espressif.com/dl/package_esp32_index.json because my seller gave it to me 1 year ago.
So i can close this issue, hoping it could help other beginners, and i ll open a new one, not properly issue.

closed 2022-05-06 by konig87nikkinakki