cotestatnt/AsyncTelegram2 #75: OTA update with ESP32

joroMaser, 2022-04-28:
I will be happy if you can an example to update FW to esp32 OTA via telegram bot
Thank you!

cotestatnt, 2022-04-29:
Here you can find what you are asking for.
The example are written for esp8266 only, but with some small edits (like in others examples) can run also with ESP32

https://github.com/cotestatnt/AsyncTelegram2/tree/master/examples/ESP8266

cotestatnt, 2022-04-29:
Ok, now the same example con compile with both ESP32 and ESP8266 MCUs

https://github.com/cotestatnt/AsyncTelegram2/tree/master/examples/ESP_OTA

cotestatnt, 2022-04-29:
It's the same. Actually default client with esp8266 it's BearSSL so it's not needed anymore to write

joroMaser, 2022-05-05:
@cotestatnt maybe it's more good to disable /remove "FLASH" button after use used thus button
Otherwise it user press few time on this button,error will appear

joroMaser, 2022-05-06:
@cotestatnt I think there is an error on OTA with esp32 
I send firmware.bin and click FLASH .. . I saw Serial logs and when I saw `Updating 491520 of 892912 bytes...` that stay there and not continue.

I test it lot of time and it's usually happend  (each time on another bytes...) when the firmware sent from telegram web and I click FLASH in telegram web

secmask, 2023-11-03:
@cotestatnt I have same issue on update OTA on ESP32, and this fix works on my ESP32 board, 

https://github.com/cotestatnt/AsyncTelegram2/pull/131

closed 2025-04-05 by cotestatnt