cotestatnt/AsyncTelegram2 #78: When message ask as read

joroMaser, 2022-05-07:
I thought that when I got new message with `getNewMessage` that message marked as read so I will not get this message again.
But I test that code to implement restart esp with text message
```
if (msg.text == "RESTART")
        {
          Serial.println("RESTART message");
          ESP.restart();
        }
```
But that makes a boot loop,each boot I got the same message(even I sent this message once).
And I saw that each time I got the same

cotestatnt, 2022-05-08:
This is the same like this https://github.com/cotestatnt/AsyncTelegram2/issues/62 and this https://github.com/cotestatnt/AsyncTelegram2/issues/70

joroMaser, 2022-05-08:
@cotestatnt That you so much about your reply. 
But using `while (!myBot.noNewMessage())` will not drop anew message after reset message (until the esp will reboot?)

cotestatnt, 2022-05-08:
`myBot.noNewMessage()` simply wait until bot is synced with server and no new messages are pending.

joroMaser, 2022-05-08:
@cotestatnt Thank you,you can close this issue.
I will check hiw do you prevent get duplicate message in normal state ` while  getNewMessage`

closed 2022-05-08 by cotestatnt