cotestatnt/AsyncTelegram2 #79: suggest a possible solution to the problem

Demon-MC, 2022-05-13:
last night one of the members of the group where the bot is installed threw a very large text into the chat
[image]
after that esp8266 hung.  there is no way to check the console.  esp8266 also stopped sending information from sensors to the Internet.
[image]
passed the last information from the esp sensors before sending a large text to the chat

question: how to protect esp from large text messages in chat so that the bot skips them and does not process them?  especially messages in Cyrillic.

Demon-MC, 2022-05-13:
forgot to write.  esp continues to work and pings over the network.  but the loop function hung at some point

joroMaser, 2022-05-14:
https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.cpp#L218-L231
Is that section didn't run? @Demon-MC

Demon-MC, 2022-05-15:
I'm not sure about this yet.. is there any way in c++ code to set the maximum length of the received command message

closed 2023-03-16 by cotestatnt