cotestatnt/AsyncTelegram2 #80: not an issue: how to get time from library?

konig87nikkinakki, 2022-06-23:
Hi Cotestatnt,
thank you again for support. i use ESP32 and i need to get time in one of my sketch. 
i ve seen there are libraries for esp32 that can read time from NTP but as i can see, your library does it while connecting to telegram server:
```
  // Sync time with NTP
  configTzTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
  #if USE_CLIENTSSL == false
    client.setCACert(telegram_cert);
  #endif
```
well, can you please tell me how to assign date and time to Variabile? after i obtain this data, what kind of library can you suggest me to get the time later, without asking NTP server each time? And where can i set Daylight and exact delay hour from greenwich?
i need to read time at begin, and later read it into a library or something without asking again NTP server.
When i work with Arduino i used DS3231 RTC and "RTClib.h" to set time or read time from RTC later.
What can i do with ESP using your library to read NTP once at start? Can i use <ESP32Time.h> ?

cotestatnt, 2022-06-23:
System time synchronization with an NTP server has nothing to do with the library. I added the code in the examples because SSL certificate verification requires time to be updated because certificates expire.

cotestatnt, 2022-06-23:
System time synchronization with an NTP server has nothing to deal with the library. I added the code in the examples because SSL certificate verification requires time to be updated because certificates expire.
You don't need external library to handle time and NTP sync with ESP32: it's all included in core code 

Check [this example](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/sendOnEvent/sendOnEvent.ino) for how you can retrieve updated time from system. The time is stored [inside the struct tm](https://m.cplusplus.com/reference/ctime/tm/) and then handled with the [standard C function strftime](https://m.cplusplus.com/reference/ctime/strftime/). You could use also printf getting each distinct member of tm struct in order to build your custom string format.

In order to set you own time zone and daylight saving time it's enought edit this line of code (mine is for Central Europe: Rome, Berlin, Amsterdam etcetc)

`#define MYTZ "CET-1CEST,M3.5.0,M10.5.0/3"`

konig87nikkinakki, 2022-06-23:
:-D i only understood #define MYTZ "CET-1CEST,M3.5.0,M10.5.0/3 , that is WARSAW, Poland. They have a good vodka andy my favorite is Zubrowka. For italy, you should use 
CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00 
but for sure Telegram Bot works because 2h difference is not a problem.
Well.
->in the Example i understood that when you push button ESP asks to NTP server about the time, am i right?
->in Struct TM i ve seen tm_hour and so forth but i dont know how to syncronize Struct TM to NTP at the begin of Sketch
->i can't see where, in the Example, you used STRFTIME.
-> why did you include time.h ?
->     time_t now = time(nullptr);  --> what does it mean nullptr?
    struct tm t = *localtime(&now); --> a pointer to localtime with the data of NOW ?

without boring you, can you link me other examples so i can understand how to use struct tm and strftime?
I d like to store Date and Time at begin, and then after some hours get it from ESP without asking to NTP again, because i'm afraid NTP server kick me off if i ask him time too often and put my ip in blacklist
many thanks, i hope this could help others

cotestatnt, 2022-06-23:
Hu @konig87nikkinakki 
Now I'm traveling and it's a little bit difficult for me give you a detailed answer.
I will as soon as can ;)

In the meantime you could check here: it use a little different approach for offset and dst setup, but mainly the principles are the same 

https://randomnerdtutorials.com/esp32-date-time-ntp-client-server-arduino/

konig87nikkinakki, 2022-06-23:
thank you again!!!
i was looking to the same example and i was editing my last Answer in this Post! 
thank you i m not in a hurry, it s a hobby, answer just if you like and if you have free time to spend!

_EDIT: i m trying to compare your sketch with this one, that "looks" easier because it doesnt log in into telegram:
https://randomnerdtutorials.com/esp32-date-time-ntp-client-server-arduino/

and i still cannot understand 
    time_t now = time(nullptr);
    struct tm t = *localtime(&now); --> to save  information about time in variable?_

cotestatnt, 2022-06-23:
```
  // The variable now contains the system time as Unix Time (number of seconds since 01/01/1970)
  time_t now = time(nullptr);
  // The instruction localtime get the value of now and fit the members of struct tm t, so t.tm_hour will be equal to actual hour, t.tm&min to minutes etcetc.
  struct tm t = *localtime(&now);
```
The function [localtime](https://www.tutorialspoint.com/c_standard_library/c_function_localtime.htm) return a pointer to struct tm, so you need to dereference the pointer returned in order to assign properly the value to the local `struct tm t`

konig87nikkinakki, 2022-06-26:
oook sorry for delay in answering to you. this morning i tested and it works fine but not for all i need, so probably i have to sue some time library external, let me explain why. CODE:

```
void funzione_orologio()
{
    // The variable now contains the system time as Unix Time (number of seconds since 01/01/1970)
  time_t now = time(nullptr);
    // The instruction localtime get the value of now
  // and fit the members of struct tm t, so t.tm_hour 
  // will be equal to actual hour, t.tm_min to minutes etcetc.
  struct tm t = *localtime(&now);
 WiFi.disconnect(true);
 WiFi.mode(WIFI_OFF);
  
delay(0);
    int i=0;
    for(i=0;i<5;i++)
    {

  Serial.println("Orologio è:");
  Serial.print(t.tm_hour);Serial.print("\t");Serial.print(t.tm_min);Serial.print("\t");Serial.println(t.tm_sec);
  int ore=0; int minuti=0; int secondi=0;
  ore=t.tm_hour; minuti=t.tm_min; secondi=t.tm_sec;
  Serial.println("mostro le variabili");
  Serial.print(ore);Serial.print("\t");Serial.print(minuti);Serial.print("\t");Serial.println(secondi);

  delay(3000);
    }
}
```

i understood that it asks the Time to NTP server when i call time_t now = time(nullptr);  , because the FOR cycle answers alway the same Hours, Mins and Seconds every time i call the funcion. i need something that get time at startup and then if i call function it goes offline with
WiFi.disconnect(true);
WiFi.mode(WIFI_OFF);
and then begin to show time every TOT seconds via Serial.
So i cannot ask NTP server time anymore except than the first time when i receive Telegram Message. i must ask NTP time to server few times because i m afraid it kick me out because of too much ask. So i must get time once at startup and then calculate in future without asking again, that s why i use WiFi.disconnect(true); and WiFi.mode(WIFI_OFF); for TEST.

cotestatnt, 2022-06-27:
> i understood that it asks the Time to NTP server when i call time_t now = time(nullptr);

No, the instruction get **system time**, not NTP time.

You will syncronize system time with an external NTP time source only during setup with the instruction 
`configTzTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");`, after that, the ESP32 will handle time keeping on it's own [using the internal RTC timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html) and for that reason it will work also without wireless connection.

konig87nikkinakki, 2022-06-27:
ok, so why if i run this Function i have alway the same time in the For cycle? in the FOR cycle i get always the same time that is the time when the cycle begin. maybe should i update the struct time? i thought i have always time updated just calling t.tm_hour , or min, or sec. i ll try again later ;-)

cotestatnt, 2022-06-27:
Because you are updating system time outside the for cycle 😉