cotestatnt/AsyncTelegram2 #81: Unstable sending of ESP32 messages

usr222, 2022-06-24:
I use the device on ESP32. I wrote a code in which, upon the occurrence of certain conditions, for example, the closure and opening of the sensor must be sent to telegrams. But I noticed that these messages are not always sent. More precisely, immediately after turning on the device for about 1 hour, everything works. Then some messages are not sent (or are they sent, but do not come?). For example, the sensor is closed - the message has gone and was received. After some time, the sensor was opened - the property came. Then, again, after some time, the sensor is closed - the message does not come. The sensor is open - the message received. And so it can happen with different events. There is no specific pattern. At the same time, the device itself works normally, without freezes, which is confirmed by duplicate messages in Serial Port. Replaced the AsyncTelegram2 library in the same code on UniversalTelegrambot - for 5 days everything has been working fine for 5 days. But since this synchronous library would like to use yours. What could be the problem?

cotestatnt, 2022-06-25:
Hi @usr222 
Could you share the part of code which is not working (or even better the complete sketch).

If you have omitted the instruction `myBot.getNewMessage(msg)`, this could be the reason because the bot need to be synced with Telegram server even if you have to "send only" messages.
Which version of library are you using?

usr222, 2022-06-25:
hi @cotestatnt 
The code is very big. Here is part of the code
`switch (mode) {
    case tmOFF :
 if (!flag_msg_off) {
Serial.println("OPEN");
bot.sendMessage(CHAT_ID, "OPEN", "");
        flag_msg_off = 1;
}
break;
case tmON :
 if (!flag_msg_on) {
Serial.println("CLOSE");
bot.sendMessage(CHAT_ID, "CLOSE", "");
        flag_msg_on = 1;
}
break;
...

}`

and yes i didn't use myBot.getNewMessage(msg). But I did not use a similar function with the UniversalTelegrambot library, but everything works there without it. I should just write at the beginning of the loop 

`TBMessage msg;
bot.getNewMessage(msg);`

 ?

cotestatnt, 2022-06-25:
UniversalTelegramBot is another library and it works in a slighlty different mode off course.
Probably what is done with AsyncTelegram2 when the method `getNewMessage()` get called is done in other way under the hood.

Yes, adding those 2 lines of code inside loop is enought and it should work without problems unitl you keep your ESP32 on and connected.

usr222, 2022-06-25:
Added. I'll watch for a few days and then report back.

usr222, 2022-06-26:
Reboots began at intervals from 10 to 120 minutes.
```
Unable to connect to Telegram server
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
 
Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060f30  A0      : 0x800e6b32  A1      : 0x3ffddf50  
A2      : 0x00000aaf  A3      : 0x00001410  A4      : 0x3ffbae10  A5      : 0x0000140b  
A6      : 0x00000001  A7      : 0x3ffdbe80  A8      : 0x800e63f8  A9      : 0x00000961  
A10     : 0x00000000  A11     : 0x00000961  A12     : 0x00000aaf  A13     : 0x3ffb3270  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000011  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400897b8  LEND    : 0x400897ce  LCOUNT  : 0xffffffff  
 
 
Backtrace:0xfffffffd:0x3ffddf500x400e6b2f:0x3ffddf70 0x400e8711:0x3ffde060 0x400e4cb8:0x3ffde110 0x400e4cdb:0x3ffde130 0x400d84ed:0x3ffde150 0x400d7e79:0x3ffde3e0 0x400d7f84:0x3ffde420 0x400d89d6:0x3ffde440 0x400d3966:0x3ffde460 0x400d3a1a:0x3ffde500 
 
 
 
 
ELF file SHA256: 0000000000000000
 
Rebooting...

```

closed 2022-07-07 by cotestatnt