cotestatnt/AsyncTelegram2 #82: Reading and processing messages from a channel or group

usr222, 2022-08-14:
Hi! There is a bot that sends messages to a channel or group. The question arose - can the second bot read and process messages from this channel/group using this library? If yes, how to do it? It worked to send a message to the channel, but not to read it using another bot. If possible, give a small example.

cotestatnt, 2022-08-24:
Hi @usr222 
Sorry for late reply, but I was on holiday.

The bot can receive and handle messages from groups, but off course **it has to be added as a member of group**.

You need only to take care about commands sent with slash character like for example **/start** or **/whateveryouwant** because when a command is sent from a group, Telegram will add automatically "@name_of_the_group" to text of command.

For example, **/start** will became **/start@<name_of_the_group>**

usr222, 2022-08-27:
Hi **@cotestatnt** 
Thanks for the answer. Yes, I do not know how, but everything has already worked. When I returned to my code a few days later, I saw that the bots were already receiving messages in the group. At the same time, I did not change anything either in my code or in the settings of the bots. Some mysticism :) . But the main problem remained - how to teach bots to see messages from other bots. As I understand it, this is disabled by the telegram developers. Perhaps there is another way to control one (master) esp8266 with other esp8266 via the Internet?

cotestatnt, 2022-08-28:
Hi @usr222 
Yes, unfortunately you can't send a message from bot to bot because, as far as i know, there's no way to get the bot's id.
If you need to make a device to device communication, you need to orient yourself on a system other than Telegram.

A possible solution could be to use Google Firebase as a "gateway": for example, the "Realtime Database" allows you to add "listeners" to the data so that every time there is a change your ESP can react accordingly without having to poll continuously.

Check [this examples included with Firebase-ESP-Client library](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/RTDB/DataChangesListener) as a stanting point!

closed 2022-09-01 by cotestatnt