cotestatnt/AsyncTelegram2 #83: Doesn't work on esp32.

zenbooster, 2022-08-19:
Hello! My application runs on ESP32 (ESP-WROOM-32), and in addition to the telegram bot, it uses a web server, and also connects to another bluetooth device in master mode.
As a result, my application does not receive messages from the getNewMessage function. Here is the log:
```
Test Telegram connection...
[D][src/AsyncTelegram2.cpp:38] checkConnection():       Start handshaking...OK
[E][BluetoothSerial.cpp:865] isReady(): BT is not initialized. Call begin() first
[E][src/AsyncTelegram2.cpp:221] getNewMessage():        deserializeJson() failed
[E][src/AsyncTelegram2.cpp:222] getNewMessage():        NoMemory
[E][src/AsyncTelegram2.cpp:223] getNewMessage():        m_rxbuffer.length()==478
[E][src/AsyncTelegram2.cpp:226] getNewMessage():        m_rxbuffer="␘��?�␁���"
The device started in master mode, make sure remote BT device is on!
[E][src/AsyncTelegram2.cpp:221] getNewMessage():        deserializeJson() failed
[E][src/AsyncTelegram2.cpp:222] getNewMessage():        NoMemory
[E][src/AsyncTelegram2.cpp:223] getNewMessage():        m_rxbuffer.length()==23
[E][src/AsyncTelegram2.cpp:226] getNewMessage():        m_rxbuffer="␘��?␗"
[E][src/AsyncTelegram2.cpp:221] getNewMessage():        deserializeJson() failed
```
Apparently the problem is in the function:
`MessageType AsyncTelegram2::getNewMessage(TBMessage &message )`

If replace:
```
            #if defined(ESP8266)
            DynamicJsonDocument updateDoc(ESP.getMaxFreeBlockSize() - BUFFER_MEDIUM);
            #elif defined(ESP32)
            DynamicJsonDocument updateDoc(ESP.getMaxAllocHeap());
            #else
            DynamicJsonDocument updateDoc(BUFFER_BIG);
            #endif

```
on:
```
            #if defined(ESP8266)
            DynamicJsonDocument updateDoc(ESP.getMaxFreeBlockSize() - BUFFER_MEDIUM);
            #elif defined(ESP32)
            DynamicJsonDocument updateDoc(BUFFER_BIG);
            #else
            DynamicJsonDocument updateDoc(BUFFER_BIG);
            #endif

```
...then everything works.
I suspect that the same should be done for the ESP8266.

zenbooster, 2022-09-15:
The "dev" branch works out of the box! )

dcrespoar, 2022-11-01:
Hello, Zenbooster the "dev" branch work because is based on version 2.0.8 where the declaration of  DynamicJsonDocument updateDoc() is like the one modified by you, DynamicJsonDocument updateDoc(BUFFER_BIG);
I am experiencing the same problem, it is definitely the DynamicJsonDocumentupdateDoc (ESP.getMaxAllocHeap()) declaration, I think it is not a good idea to use such amount of memory in a multitasking environment.

closed 2023-03-16 by cotestatnt