cotestatnt/AsyncTelegram2 #85: Caption

navdeepssidhu, 2022-08-28:
Hi, thanks for providing excellent Lib. This is not an issue only a request. Is it possible to add caption to picture sent to telegram bot?
Thanks

navdeepssidhu, 2022-08-28:
or show timestamp on picture

navdeepssidhu, 2022-08-28:
or is it possible to show time stamp on the picture

cotestatnt, 2022-08-28:
HI @navdeepssidhu
At the moment you can send a picture with caption only from external url, but it is not a limitation of library itself, only my forgetfulness.

The needs of users who use a Telegram bot are many and it is difficult to predict them all from the beginning.
For this reason, the Github issues are a valuable source of feedback and suggestions for which I thank you.

I will add the ability to add a caption to the picture (which can therefore also be a timestamp) as soon as I have some time to dedicate to the library!

cotestatnt, 2022-08-28:
@navdeepssidhu it was easier than I thought because I had already foreseen the possibility, but leaved unimplemented. It was enough to make the caption an option parameter for **sendPhoto()** methods.

On this occasion I updated the [sendPhoto.ino example](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/sendPhoto/sendPhoto.ino#L210).
You can find the changes in latest release 2.1.4

navdeepssidhu, 2022-08-28:
Thank you so much for your quick work. I will give it a go. Is a timestamp on picture itself possible?

cotestatnt, 2022-08-28:
If you intend to merge the timestamp with the image, no it's not possible just using the Telegram API.

You have to use caption field for this purpose.

```C++
   ....
  time_t now = time(nullptr);
  struct tm sysTime= *localtime(&now);
  char timestamp[20];

  // https://cplusplus.com/reference/ctime/strftime/
  strftime(timestamp, sizeof(timestamp), "%d/%m/%Y - %T", &sysTime);

  // sendPhoto(const TBMessage &msg, const char* filename, fs::FS &fs, const char* caption = nullptr)
  myBot.sendPhoto(msg, "/telegram-bot1.jpg", FILESYSTEM, timestamp);
  ...
```

navdeepssidhu, 2022-08-29:
Thanks a ton for updating the example.

closed 2022-08-29 by navdeepssidhu