cotestatnt/AsyncTelegram2 #86: Markdown to channel

okruts, 2022-09-05:
Is it possible to send markdown message to a channel? I didn't find that option with `sendToChannel` method

cotestatnt, 2022-09-05:
Hi @okruts, markdown message don't rely on which methos you use in order to send message, but on message itself.
It's a property of the [TBMessage struct](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/DataStructures.h#L51) and by default it's enabled [HTML markdown fromatting](https://core.telegram.org/bots/api#formatting-options) because I think it is more flexible and it is not confusing if you have to use one of the characters used by markdown.

If you want to use MarkdownV2 you have to set it manually before sending the message.

```
msg.isMarkdownEnabled  = true;
myBot.sendMessage(msg, "Send a markdown formatted message: *bold \*text*"); 
```

For the future I am thinking of adding a configuration method to set default formatting option in setup or when needed.

okruts, 2022-09-05:
@cotestatnt Yes I saw that. But it is not possible to pass msg argument when you specifically want to send message to public channel using `myBot.sendToChannel`, not to user or group: https://github.com/cotestatnt/AsyncTelegram2/blob/a01a749af6dadd57102b5cb6fdd5a0d1f3d14457/src/AsyncTelegram2.h#L175 
I tried to use `myBot.sendMessage` method to send messages to channel using channel id but it doesn't work.

cotestatnt, 2022-09-06:
Ops, you are right.
I forgot that the way to send messages to a channel is slightly different.

Unfortunately in this way it is not possible to use the formatting options, I have to correct this lack as soon as possible and i think the easiest way is to add a parameter to `sendToChannel()` as you suggested.

cotestatnt, 2022-09-09:
Hi @okruts 
I've just drafted a new release with a new global method which allows you to set the format style (none, HTML, MarkdownV2) for any type of message sent.

Check this new example for hot-to use it and please let me know about any issue or advice.
https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/formatStyle/formatStyle.ino

closed 2022-09-15 by cotestatnt