cotestatnt/AsyncTelegram2 #87: Clear() method

xlettera, 2022-09-10:
Dear Tolentino,
First of all thank you for having added clear method.
Maybe I am using it in a wrong way, or I am doing something wrong. But the keyboard does not show up.

The general idea is to define **just one keyboard** with the reply keyboard object helper, then with clear method I would like to empty it and then I would like to redefine and reuse it with another layout or with other labels for example.

I have modified your **keyboards** sketch to show you this issue. 
If you remove _myReplyKbd.clear();_  from the code and you type _/reply_keyboard1_ you will get keyboard1 but if you type _/reply_keyboard2_ you will have two keyboards together. I you put _myReplyKbd.clear();_ the keyboard does not show up.
Only the right keyboard should appear on the screen.

I hope I was clear.  
Thank you for your help.
Roberto

```
/*
  Name:        keyboards.ino
  Created:     05/03/2022
  Author:      Tolentino Cotesta <cotestatnt@yahoo.com>
  Description: a more complex example that do:
             1) if a "/inline_keyboard" text message is received, show the inline custom keyboard,
                if a "/reply_keyboard" text message is received, show the reply custom keyboard,
                otherwise reply the sender with "Try /reply_keyboard or /inline_keyboard" message
             2) if "LIGHT ON" inline keyboard button is pressed turn on the LED and show a message
             3) if "LIGHT OFF" inline keyboard button is pressed, turn off the LED and show a message
             4) if "GitHub" inline keyboard button is pressed,
                open a browser window with URL "https://github.com/cotestatnt/AsyncTelegram"
*/
#include <AsyncTelegram2.h>

// Timezone definition
#include <time.h>
#define MYTZ "CET-1CEST,M3.5.0,M10.5.0/3"

/*
  Set true if you want use external library for SSL connection 
  For example https://github.com/OPEnSLab-OSU/SSLClient/ is very efficient BearSSL library and 
  you can use AsyncTelegram2 even with other MCUs or transport layer (ex. Ethernet or GPRS)
*/
#define USE_CLIENTSSL false

#ifdef ESP8266
  #include <ESP8266WiFi.h>
  WiFiClientSecure client;
  Session   session;
  X509List  certificate(telegram_cert);
#elif defined(ESP32)
  #include <WiFi.h>
  #include <WiFiClient.h>
  #if USE_CLIENTSSL
    #include <SSLClient.h>      // https://github.com/OPEnSLab-OSU/SSLClient/
    #include "tg_certificate.h"
    WiFiClient base_client;
    SSLClient client(base_client, TAs, (size_t)TAs_NUM, A0, 1, SSLClient::SSL_ERROR);
  #else
    #include <WiFiClientSecure.h>
    WiFiClientSecure client;
  #endif
#endif

AsyncTelegram2 myBot(client);
const char* ssid  =  "xxxxxxxxx";     // SSID WiFi network
const char* pass  =  "xxxxxxxxx";     // Password  WiFi network
const char* token =  "xxxxxxxxxxx:xxxxxxxxxxxxxxxxxxxxxxxxxxxxx";  // Telegram token

// Check the userid with the help of bot @JsonDumpBot or @getidsbot (work also with groups)
// https://t.me/JsonDumpBot  or  https://t.me/getidsbot
int64_t userid = 123456789;

ReplyKeyboard myReplyKbd;   // reply keyboard object helper
[... 193 lines cut ...]
          reply += msg.location.latitude;
          Serial.println(reply);
          myBot.sendMessage(msg, reply);
          break;
        }

      case MessageContact: {
          // received a contact message
          String reply = "Contact information received:";
          reply += msg.contact.firstName;
          reply += " ";
          reply += msg.contact.lastName;
          reply += ", mobile ";
          reply += msg.contact.phoneNumber;
          Serial.println(reply);
          myBot.sendMessage(msg, reply);
          break;
        }

      default:
        break;
    }
  }
}
```

cotestatnt, 2022-09-11:
ok, thanks for your valuable feedback. I will do some more in-depth tests as soon as possible so that I have a clear idea of ​​what is going on

xlettera, 2022-09-12:
Perfect. 
This will be a big step forward for this library.
Regards
Roberto

cotestatnt, 2022-09-12:
So, I checked and actually I made a mistake in resetting the variable. 
I had to reinitialize using an empty - but valid - keyboard JSON instead using only an empty string.

However, I still believe this way of proceeding is not the best possible.
I would make sure that the functions `defineMyKbdStart()` and `defineMyKbdOption()` return a `String` so as to be able to use a local variable only when needed and avoid the risk of causing excessive fragmentation of the heap (or you can pass to the functions the reference to a ReplyKeyboard object as I suggested in the other answer).

```C++
String defineMyKbdStart()
{
  isKeyboardActive = false;

  ReplyKeyboard kbd;
  // add a button that send a message with "Simple button" text
  kbd.addButton("Button1");
  kbd.addButton("Button2");
  kbd.addButton("Button3");
  // add a new empty button row
  kbd.addRow();
  // add another button that send the user position (location)
  kbd.addButton("Send Location", KeyboardButtonLocation);
  // add another button that send the user contact
  kbd.addButton("Send contact", KeyboardButtonContact);
  // add a new empty button row
  kbd.addRow();
  // add a button that send a message with "Hide replyKeyboard" text
  // (it will be used to hide the reply keyboard)
  kbd.addButton("/hide_keyboard");
  // resize the keyboard to fit only the needed space
  kbd.enableResize();
  return kbd.getJSON();
}

String defineMyKbdOption()
{
  isKeyboardActive = false;

  ReplyKeyboard kbd;
  // add a button that send a message with "Simple button" text
  kbd.addButton("Option1");
  kbd.addButton("Option2");
  kbd.addButton("Option3");
  // add a new empty button row
  kbd.addRow();
  kbd.addButton("/hide_keyboard");
  // resize the keyboard to fit only the needed space
  kbd.enableResize();
  return kbd.getJSON();
}
```

```C++
       if (msgText.equalsIgnoreCase("/reply_keyboard1")) {

          // the user is asking to show the reply keyboard --> show it
          myBot.sendMessage(msg, "This is reply keyboard1:", defineMyKbdStart());
          isKeyboardActive = true;
        }
        else if (msgText.equalsIgnoreCase("/reply_keyboard2")) {

          // the user is asking to show the reply keyboard --> show it
          myBot.sendMessage(msg, "This is reply keyboard2:", defineMyKbdOption());
          isKeyboardActive = true;
        }
```

xlettera, 2022-09-12:
Dear Tolentino,
The clear() method now works perfectly.
I will follow your advices. 
Thank you.
Regards
Roberto

closed 2022-09-12 by xlettera