cotestatnt/AsyncTelegram2 #88: My project doesnt work anymore when I compile and upload it using the updated library

FBMinis, 2022-09-12:
One year ago I put together the following project:

https://github.com/FBMinis/DS3231-Alarm-ESP32-Cam-Telegram-Bot

After updating your library to 2.1.5 the program compiles and uploads but I never receive the photo on Telegram.

It worked on: Windows 10; Arduino portable 1.8.13; ESP32 2.0.0; AsyncTelegram2 2.0.4

Currently not working on: Win10; Arduino portable 1.8.19; ESP32 2.0.4; AsyncTelegram2 2.1.5

The example "ESP32-CAM" included in your library works fine.

My interpretation is that something changed since 2.0.4 regarding the way the photo info is sent to the server. I read your documentation but have not been able to identify what might be causing this situation.

I would appreciated some suggestions from you on how to solve this. Thank you!

cotestatnt, 2022-09-12:
Hi @FBMinis, first of all thank you for the feedback which is always very useful for finding insidious bugs like this.

The way the photo is sent to the server should not be changed a lot... in the last releases I've only added the possibility to include caption with pictures.

I did my tests by sending images directly from the flash, but unfortunately not using an ESP32-CAM board.
As soon as I can I try to verify your sketch to see if I can identify the problem with the new version of the library.

FBMinis, 2022-09-12:
My project was based on your "ESP32-CAM" example, it sends a photo and goes to deep sleep for a defined interval.

I compared the example in the 2.0.4 version with the one in the 2.1.5 version. I commented out the lines that appeared different and it's working normally now:

```
//#define USE_SSLCLIENT false
//#if USE_SSLCLIENT
//#include <SSLClient.h>
//#include "tg_certificate.h"
//WiFiClient base_client;
//SSLClient client(base_client, TAs, (size_t)TAs_NUM, A0);
//#else
#include <WiFiClientSecure.h>
WiFiClientSecure client;
//#endif
```

```
//#if USE_CLIENTSSL == false
  client.setCACert(telegram_cert);
//#endif
```

closed 2022-09-12 by FBMinis

cotestatnt, 2022-09-13:
Hi @FBMinis 

The problem related to your code should be due to this line `msg.sender.id = userid;`
Unfortunately this data structure can lead to some confusion as at some point I decided to treat the sending of messages to a specific user or group in the same way.

In fact the only difference is the ID which in the case of groups is in negative int64.
If you modify the sendPicture () function in this way it should also work with the latest version of the library.

FBMinis, 2022-09-13:
I made the changes you suggested and it's working again, thank you!