cotestatnt/AsyncTelegram2 #89: cannot connect to telegram server if used with sqlite3 for arduino

sharkyenergy, 2022-11-19:
Hello!

I have the wifi data stored on a SD card inside a sqlite database.
In order to connect to my wifi, i read that data from the database using the SQLITE3 for arduino library (https://github.com/siara-cc/esp32_arduino_sqlite3_lib)

The problem is that as soon as I run this:

    rc = sqlite3_prepare_v2(db1, sql.c_str(), 1000, &res, &tail);

(sample code here: https://github.com/siara-cc/esp32_arduino_sqlite3_lib/blob/master/examples/sqlite3_webquery/sqlite3_webquery.ino)

your library stops working.. I cannot connect to telegram anymore. I get the error message "Unable to connect to Telegram, NOK" .

I am not able to figure out why.. what is going on here?

Any help is greatly appreciated.

best regards

sharkyenergy, 2022-11-20:
is there any debugging I can enable that tells me more than "unable to connect"? to figure out what the cause is? thank you

cotestatnt, 2023-03-16:
Hi @sharkyenergy, I'm sorry for the big late reply, but I've been very busy in recent months.

Do you still have this problem or did you manage to solve it?

closed 2025-04-05 by cotestatnt